projecthorus/radiosonde_auto_rx #770: auto_rx will fail if started and RTL-SDR are already initialised

myhillman, 2023-05-11:
In a configuration where the RTL-SDR are powered by an external USB hub, I suspect the auto_rx service will fail on startup because the RTL-SDR are not in the Power On Reset state. In such situations, when you cycle the power on the RPi (but not the RTL-SDR), the system will not return to an operational state.

Seems that the code assumes that the RTL-SDR are in the Power On Reset state at startup. This may not necessarily be the case. Software should not make this assumption and reset the RTL-SDR before startup.

This is speculation based on the observed behaviour of the device in the specified configuration.

Affected versions 1.6.0 & 1.6.1.

This is low priority as it only affects installations using externally powered USB hubs, and there is a work-around (power cycle hub as well as RPi)

darksidelemm, 2023-05-11:
What is the behaviour of auto_rx in this state? I would need to see a detailed log of how it breaks on startup to figure out if there is anything I can do.

If it's not even detecting the RTLSDRs then I'm not sure what I can do about it.

myhillman, 2023-05-11:
The observed symptom is that auto_rx is not running. Local access to http://localhost:5000 returns access is denied.

Unfortunately my Linux skills are so poor I don't know where to find a log.

darksidelemm, 2023-05-11:
If you are running auto_rx as a 'native' install (using systemd), then you can get the last few hundred lines of log output by running: 
```
sudo journalctl -u auto_rx.service -n 200
```
This should hopefully include any errors occurring on startup.

myhillman, 2023-05-11:
Don't know what I'm running. I used Docker to install on a clean SD card. I have Putty and access via FileZilla (SFTP)

darksidelemm, 2023-05-11:
OK, if you're using docker, then you can run the command:
```
docker logs --tail 200 radiosonde_auto_rx
```
This will give you the log output which will hopefully provide error messages.

myhillman, 2023-05-11:
Argh! I can't reproduce the fault now. Tried 3 times and it starts reliably even if the SDR remain powered on. As it can't be reproduced, I suggest you close it. I will grab a log if it happens again.

myhillman, 2023-05-11:
Not reproducible

closed 2023-05-11 by myhillman