projecthorus/radiosonde_auto_rx #771: dont see spectra while RTLSDR scanning

laarhoven1, 2023-05-13:
spectra peaks not visible on scanpot

laarhoven1, 2023-05-13:
Problem found its not showing in firefox but with google crome its ok

darksidelemm, 2023-05-13:
Can you please look in your browser console for any errors?

darksidelemm, 2023-05-13:
Also, a screenshot showing the issue (perhaps the same scan plot shown on both firefox and chrome) would be handy...

darksidelemm, 2023-05-13:
This is what I see on Firefox on my machine. Appears to be working here :-/

[image]

laarhoven1, 2023-05-14:
[image]
[image]
First a had this issue after rechecking with firefox today the problem seems to be solved

closed 2023-05-14 by laarhoven1