projecthorus/radiosonde_auto_rx #773: FEATURE REQUEST: Imperial Altitude

vk2mpj, 2023-05-16:
Hi,
Would it be possible to enable users to select individual measurements to be displayed in imperial / metric units.
The current settings of metric / imperial changes all measurements across the board. What I would like to do is display altitude in ft and alt rate in ft/s, but leave temp, velocity and range in their respective metric units.

Thanks for all the great work and ongoing development

de 
VK2MPJ

darksidelemm, 2023-05-17:
I will say I'm a little bit loathe to implement this. It's not going to be high on my priorities. Adding imperial support at all was only done because someone submitted a PR.

vk2mpj, 2023-05-17:
> I will say I'm a little bit loathe to implement this. It's not going to be high on my priorities. Adding imperial support at all was only done because someone submitted a PR.

Ok fair enough, Ill have a look around and see if I can work something out locally, if I can find the relevant config files

vk2mpj, 2023-05-18:
Ok, I have edited index.html and it looks like it'll do what I am wanting. I'll have to wait until tomorrow for the next balloon launch to see if it works. If it does, I'll make the edited file available for anyone else who wants to do the same.

pinkavaj, 2024-05-24:
Did it worked for you? @vk2mpj

closed 2024-11-16 by darksidelemm