projecthorus/radiosonde_auto_rx #778: Web UI not usable when behind a reverse proxy

buttim, 2023-06-03:
All references to resources in the web UI are absolute (e.g. /static/js/jquery-3.6.0.min.js](http://176.65.85.137:1994/static/js/jquery-3.6.0.min.js). If you configure a reverse proxy to protect access to the  web server you are forced to use (a different port) and a path of /, otherwise all references to resources will be broken.
I think it should be quite easy to just drop the leading / to permit use of an arbitrary path in the reverse proxy

darksidelemm, 2023-06-03:
So the URLS are generated using flask's templating engine, e.g. see here:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/templates/index.html#L19

```
<script src="{{ url_for('static', filename='js/jquery-3.6.0.min.js') }}"></script>
```
Flask replaces the bit in {{}}'s with the correct path. 

It might be possible to just replace those lines with (for example) `static/js/filename.js`? I haven't tried.

buttim, 2023-06-04:
I don't know a think about flasky, so, yes, that might solve the problem. But I don't know it it's the correct way to fix it.
I do suspect though that assigning a different value to what seems the value of `static` would be a cleaner solution. Am I wrong?

PR #789 (2023-07-29): Fix reverse proxy path handling
Fixes #778
closed 2023-08-06 by darksidelemm