projecthorus/radiosonde_auto_rx #779: Radisonde Auto_RX sdrs disappear on the web interface

Mopar44084, 2023-06-07:
after awhile both of my nooelec sdrs disappear on the web interface where it says Task:
I have one sdr serialized at 00000403 and the other at 00000978. 

here is my sdr settings:
################
# SDR SETTINGS #
################
[sdr]

#
# SDR Type
#
# RTLSDR - Use one or more RTLSDRs
#
# EXPERIMENTAL / NOT IMPLEMENTED options:
# SpyServer - Use an Airspy SpyServer
# KA9Q - Use a KA9Q SDR Server (Not yet implemented)
# WARNING: These are still under development and may not work.
#
sdr_type = RTLSDR

#
# Number of SDRs or SDR Connection Threads to use
#
# If SDR type is set to RTLSDR above, then this number is the number of individual RTLSDRs
# that will be used, eith each RTLSDR allocated a scan or decode task.
# If more than one RTLSDR is in use, multiple [sdr_X] sections must be populated below
#
# If SDR type is either KA9Q or SpyServer, this defines the maximum number of parallel
# decoding/scan tasks. On a RPi 4, ~5 tasks are possible.
#
sdr_quantity = 2

#
# Network SDR Connection Details
#
# If using either a KA9Q or SpyServer network server, the hostname and port
# of the server needs to be defined below. Usually this will be running on the
# same machine as auto_rx, so the defaults are usually fine.
#
sdr_hostname = localhost
sdr_port = 5555

#
# Individual RTLSDR Settings
#
# Provide details of your RTLSDRs here, e.g. device numbers, gain settings, and PPM offsets.
# If using a Network SDR, do not modify any of these settings.
#
[sdr_1]
# Device Index / Serial
#       If using a single RTLSDR, set this value to 0
#       If using multiple SDRs, you MUST allocate each SDR a unique serial number using rtl_eeprom
#   The serial number must NOT be 00000000 or 00000001, as this causes big confusion to the rtl utilities.
#       i.e. to set the serial number of a (single) connected RTLSDR: rtl_eeprom -s 00000002
#       Then set the device_idx below to 00000002, and repeat for the other [sdr_n] sections below
#
#   For the special case when auto_rx is used entirely with a JSON-via-UDP input, set the below to TCP001
#   This will bypass all RTLSDR checks and allow auto_rx to continue running.
device_idx = 00000403

# Frequency Correction (ppm offset)
#       Refer here for a method of determining this correction using LTE signals: https://gist.github.com/darksidelemm/b517e6a9b821c50c170f1b9b7d65b824
ppm = 0

# SDR Gain Setting
#   Gain settings can generally range between 0dB and 40dB depending on the tuner in use.
#   Run rtl_test to confirm what gain settings are available, or use a value of -1 to use hardware automatic gain control (AGC).
#   Note that this is an overall gain value, not an individual mixer/tuner gain. This is a limitation of the rtl_power/rtl_fm utils.
gain = -1

# Bias Tee - Enable the bias tee in the RTLSDR v3 Dongles.
bias = False

[sdr_2]
# As above, for the next SDR, if used. Note the warning about serial numbers.
device_idx = 00000978
ppm = 0
gain = -1
bias = False

# Add more SDR definitions here if needed. ([sdr_3], [sdr_4])

darksidelemm, 2023-06-07:
The settings don't really help me. You are going to have to look at the logs produced by auto_rx to look for any errors.

Mopar44084, 2023-06-09:
Where are the log files usually located?

darksidelemm, 2023-06-30:
Sorry, missed this email. Depending on how you are running auto_rx you'll need to either look in /var/log/syslog using:
```
sudo cat /var/log/syslog | grep auto_rx
```

or if using docker, using:
```
docker logs radiosonde_auto_rx
```

Mopar44084, 2023-07-01:
What happens is it scans on multiple frequencies then it seems to be stuck on one. here is my logs:

2023-07-01 02:38:13,968 INFO:Reading configuration file...
2023-07-01 02:38:13,975 WARNING:Config - Web Password not set, disabling web control
2023-07-01 02:38:13,975 WARNING:Please do not use APRS ports which forward data out to the wider APRS-IS network and cause network congestion. Switching to default port of 14590. If you believe this to be in error, please raise an issue at https://github.com/projecthorus/radiosonde_auto_rx/issues
2023-07-01 02:38:15,989 INFO:Config - Tested RTLSDR #00000403 OK
2023-07-01 02:38:17,965 INFO:Config - Tested RTLSDR #00000978 OK
2023-07-01 02:38:17,967 INFO:Started Flask server on http://0.0.0.0:5000
2023-07-01 02:38:17,969 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2023-07-01 02:38:17,970 INFO:OziMux - Started OziMux / Payload Summary Exporter
2023-07-01 02:38:17,974 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2023-07-01 02:38:18,490 INFO:Version - Local Version: 1.6.1 - Up to date!
2023-07-01 02:38:18,490 INFO:Task Manager - SDR #00000403 has been allocated to Scanner.
2023-07-01 02:38:18,951 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2023-07-01 02:38:20,457 INFO:Scanner (RTLSDR 00000403) - Starting Scanner Thread
2023-07-01 02:38:20,459 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:38:44,898 INFO:Reading configuration file...
2023-07-01 02:38:44,907 WARNING:Config - Web Password not set, disabling web control
2023-07-01 02:38:44,908 WARNING:Please do not use APRS ports which forward data out to the wider APRS-IS network and cause network congestion. Switching to default port of 14590. If you believe this to be in error, please raise an issue at https://github.com/projecthorus/radiosonde_auto_rx/issues
2023-07-01 02:38:46,955 INFO:Config - Tested RTLSDR #00000403 OK
2023-07-01 02:38:48,939 INFO:Config - Tested RTLSDR #00000978 OK
2023-07-01 02:38:48,948 INFO:Started Flask server on http://0.0.0.0:5000
2023-07-01 02:38:48,951 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2023-07-01 02:38:48,955 INFO:OziMux - Started OziMux / Payload Summary Exporter
2023-07-01 02:38:48,959 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2023-07-01 02:38:51,333 INFO:Version - Local Version: 1.6.1 - Up to date!
2023-07-01 02:38:51,334 INFO:Task Manager - SDR #00000403 has been allocated to Scanner.
2023-07-01 02:38:51,543 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2023-07-01 02:38:53,291 INFO:Scanner (RTLSDR 00000403) - Starting Scanner Thread
2023-07-01 02:38:53,292 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:39:15,242 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:39:32,124 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:39:52,238 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:40:09,141 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:40:29,245 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2023-07-01 02:40:39,057 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2023-07-01 02:40:40,336 INFO:Flask - New Web Client connected!
2023-07-01 02:40:46,151 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:41:06,164 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 10 frequencies (MHz): [403.2  402.96 402.97 405.28 405.29 402.98 405.27 402.99 402.95 405.3 ]
2023-07-01 02:41:15,880 INFO:Flask - New Web Client connected!
2023-07-01 02:42:25,117 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:42:38,364 INFO:Flask - New Web Client connected!
2023-07-01 02:42:45,173 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 5 frequencies (MHz): [403.2  405.28 405.29 405.27 405.3 ]
2023-07-01 02:43:22,733 INFO:Flask - New Web Client connected!
2023-07-01 02:43:29,732 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:43:50,210 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:44:07,112 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:44:27,198 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:44:44,086 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:44:53,584 INFO:Flask - New Web Client connected!
2023-07-01 02:45:04,211 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:45:21,114 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.
2023-07-01 02:45:41,293 INFO:Scanner (RTLSDR 00000403) - Detected peaks on 1 frequencies (MHz): [403.2]
2023-07-01 02:45:58,200 INFO:Scanner (RTLSDR 00000403) - Running frequency scan.

darksidelemm, 2023-07-01:
This is just auto_rx reporting that it scanning, and it has found a peak on 403.2 MHz. No radiosonde is detected, so the scan keeps running, as is expected.
403.2 MHz is a known spur within RTLSDRs, so you may wish to add this to your 'never_scan' list here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L127

Back to your original issue, this is most like your RTLSDRs 'breaking' after a while. This is a known issue with RTLSDRs on platforms like the RPi and other SBCs, and can be related to USB bandwidth or power issues, with the only resolution being to power cycle. Not something I can solve unfortunately :-(

Mopar44084, 2023-07-01:
Ok thanks! I am running it on a Pi-400 32 bit Raspbian with a USB power hub

closed 2023-08-06 by darksidelemm