projecthorus/radiosonde_auto_rx #794: Flask/werkzeug Related Error When Upgrading to 1.6.2

ZigiWalter, 2023-08-04:
Hi,

I use a native installation, and followed the notes in the release notes. Strangely, I'm still unable to run radiosonde_auto_rx.

Here is what I do:

`pi@RPI3B:~/installs/myAutoRX/auto_rx $ sudo apt-get remove python3-flask
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'python3-flask' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  arduino-core avr-libc avrdude binutils-avr extra-xdg-menus gcc-avr libftdi1 libjna-java libjna-jni libllvm6.0 librxtx-java python3-click python3-colorama python3-itsdangerous
  python3-jinja2 python3-markupsafe python3-pyinotify python3-pyperclip python3-simplejson rpi.gpio-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@RPI3B:~/installs/myAutoRX/auto_rx $ sudo pip3 install flask
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: flask in /usr/local/lib/python3.7/dist-packages (2.2.5)
Requirement already satisfied: Werkzeug>=2.2.2 in /usr/local/lib/python3.7/dist-packages (from flask) (2.2.3)
Requirement already satisfied: importlib-metadata>=3.6.0; python_version < "3.10" in /usr/local/lib/python3.7/dist-packages (from flask) (6.7.0)
Requirement already satisfied: Jinja2>=3.0 in /usr/local/lib/python3.7/dist-packages (from flask) (3.1.2)
Requirement already satisfied: itsdangerous>=2.0 in /usr/local/lib/python3.7/dist-packages (from flask) (2.1.2)
Requirement already satisfied: click>=8.0 in /usr/local/lib/python3.7/dist-packages (from flask) (8.1.6)
Requirement already satisfied: MarkupSafe>=2.1.1 in /usr/local/lib/python3.7/dist-packages (from Werkzeug>=2.2.2->flask) (2.1.3)
Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.7/dist-packages (from importlib-metadata>=3.6.0; python_version < "3.10"->flask) (3.15.0)
Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in /usr/local/lib/python3.7/dist-packages (from importlib-metadata>=3.6.0; python_version < "3.10"->flask) (4.7.1)`

And this is the error I get when running:
`pi@RPI3B:~/installs/myAutoRX/auto_rx $ python3 /home/pi/installs/myAutoRX/auto_rx/auto_rx.py --config station.cfg
Traceback (most recent call last):
  File "/home/pi/installs/myAutoRX/auto_rx/auto_rx.py", line 55, in <module>
    from autorx.web import (
  File "/home/pi/installs/myAutoRX/auto_rx/autorx/web.py", line 29, in <module>
    from flask_socketio import SocketIO
  File "/usr/local/lib/python3.7/dist-packages/flask_socketio/__init__.py", line 24, in <module>
    from werkzeug.serving import run_with_reloader
ImportError: cannot import name 'run_with_reloader' from 'werkzeug.serving' (/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py)
`

What am I doing wrong...?

Thanks.

darksidelemm, 2023-08-04:
It looks like you've already got flask installed via pip.

Try:
`sudo pip3 install -U flask`

ZigiWalter, 2023-08-04:
Thanks - but still the same:
[image]

darksidelemm, 2023-08-04:
OK, can you also try:
```
sudo pip3 install -U flask-socketio
```
It looks like both need to be upgraded.

ON4CDJ, 2023-08-04:
Same error here after upgrading to 1.6.2.
Got the error 'ModuleNotFoundError: No module named 'werkzeug.middleware', and did as discribed 'remove python3-flask' and then the command 'sudo pip3 install flask'.
Rebooted and started with 'python3 auto_rx.py' as usual, but got the same error as ZigiWalter.
Also tried the suggestion by darksidelemm, but same error afterwards.

Patrick

ZigiWalter, 2023-08-04:
Much better!

Thanks for the quick help!

darksidelemm, 2023-08-04:
@ON4CDJ  Please try updating flask-socketio as well, which seems to fix this.

I've updated the release notes to what needs to be done. Thanks for bringing this to my attention!

ON4CDJ, 2023-08-04:
> @ON4CDJ Please try updating flask-socketio as well, which seems to fix this.
> 
> I've updated the release notes to what needs to be done. Thanks for bringing this to my attention!

Fixed !
1.6.2 is running now....thanks !

darksidelemm, 2023-08-04:
Excellent, great to hear! There's always a few teething issues when I need to use newer library versions.
I'll close this issue now.

closed 2023-08-04 by darksidelemm