projecthorus/radiosonde_auto_rx #800: Sdr-rtl v4

ly5p, 2023-08-19:
Rtl-sdr.com are now selling new v4 dongles with different chip. Is there any known issues? Doea it still has same sprikes? Will the install instructions be adapted to use with v4?

Thanks.

darksidelemm, 2023-08-19:
It doesn't look like the v4 RTLSDR Is going to offer any improvement over the v3 RTLSDR for radiosonde chasing. In fact, it looks like the performance will be slightly degraded, so I wouldn't recommend it. 
Also, it's only a limited-time thing, not a proper replacement for the v3. 

Unfortunately their requirement to need a different set of libraries to drive it makes things somewhat annoying for us, as we would have to switch to their fork of the rtlsdr libraries.

helioshk, 2023-09-08:
Surprisingly the v4 worked with the old drivers for me (debian, rtl-sdr 0.6.0-4). But I still installed the new drivers.

400 MHz is one of the few frequencies where the v4 has a better MDS than the v3 according to their [chart](https://www.rtl-sdr.com/wp-content/uploads/2023/08/MDS_v3_vs_v4_compare-1024x528.png), but perhaps within measurement accuracy

darksidelemm, 2023-09-08:
I've been told the v4 rtlsdr will show up and run without any critical errors, but the tuner doesn't get set correctly so you don't see anything.

Anyway, the changes required to support the v4 RTLSDR are in the upstream librtlsdr repository that we use in our docker images, so at the very least the testing branch docker image should support the v4 RTLSDR now.

darksidelemm, 2023-10-26:
Closing this as there's support in the docker image, and if you follow the guide and compile the rtlsdr utilities from source, it should also work.

closed 2023-10-26 by darksidelemm