projecthorus/radiosonde_auto_rx #802: spyserver multi tracking does not work with 1.6.2

gmke, 2023-08-29:
I had tremendous success with 1.6.1 and spyserver with my Airspy Mini, receiving and decoding up to seven sondes. Yesterday I upgraded to 1.6.2 (service stopped, spyserver killed, git pull, make, service start, spyserver start) and unfortunately it does not work anymore. Detecting the sondes still works but decoding stops after a second and the sonde never gets listed and displayed. 

Markus

darksidelemm, 2023-08-29:
Please check your logs for any errors.

There are many stations running 1.6.2 with spyserver, so I suspect this is a local issue, perhaps some dependency issue.

gmke, 2023-08-29:
thanks for the hint. Was a stupid update error by myself. Should have run ./build.sh and not just make. Closed.

Markus

closed 2023-08-29 by gmke