projecthorus/radiosonde_auto_rx #813: ERROR:Exception index 926 is out of bounds for axis 0 with size 920

ON4CDJ, 2023-09-24:
Lately I experienced this error almost everytime when I go into the Sonde List History and select one or more sondes and click on the 'Plot Path' button, nothing appears on the map, the buttons go all dark and nothing can be done, only when I refresh the page the buttons are clickable again.
The error in the terminal window is as follows:

2023-09-24 09:59:28,896 ERROR:Exception index 926 is out of bounds for axis 0 with size 920 in calculate_skewt_data
Traceback (most recent call last):
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/log_files.py", line 397, in calculate_skewt_data
    if temperature[i] < -260.0:
IndexError: index 926 is out of bounds for axis 0 with size 920
--- Logging error ---
Traceback (most recent call last):
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/log_files.py", line 476, in read_log_by_serial
    data = read_log_file(_matching_files[0], skewt_decimation=skewt_decimation)
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/log_files.py", line 354, in read_log_file
    decimation=skewt_decimation,
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/log_files.py", line 397, in calculate_skewt_data
    if temperature[i] < -260.0:
IndexError: index 926 is out of bounds for axis 0 with size 920

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/logging/__init__.py", line 1034, in emit
    msg = self.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 880, in format
    return fmt.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 619, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.7/logging/__init__.py", line 380, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/usr/lib/python3.7/threading.py", line 885, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 361, in handle
    super().handle()
  File "/usr/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 333, in run_wsgi
    execute(self.server.app)
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 320, in execute
    application_iter = app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2552, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask_socketio/__init__.py", line 44, in __call__
    start_response)
  File "/usr/local/lib/python3.7/dist-packages/engineio/middleware.py", line 74, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/middleware/proxy_fix.py", line 187, in __call__
    return self.app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2529, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/web.py", line 310, in flask_get_log_by_serial
    return json.dumps(read_log_by_serial(serial))
  File "/home/pi/HAM/radiosonde_auto_rx/auto_rx/autorx/log_files.py", line 479, in read_log_by_serial
    logging.exception(f"Error reading file for serial: {serial}", e)
Message: 'Error reading file for serial: U4574918'
Arguments: (IndexError('index 926 is out of bounds for axis 0 with size 920'),)

Any thoughts ?

Tnx, 73 !
Patrick

darksidelemm, 2023-09-24:
Yes there seems to be some issue in the calculations. I don't know what the issue is yet, but I have an example file which causes the issue so I should be able to figure it out when I get time.

darksidelemm, 2023-09-24:
I've got a fix for this in the testing branch. There was a bug in how the data was decimated for plotting.
Will leave this open until it's been merged into the master branch.

ON4CDJ, 2023-09-25:
> I've got a fix for this in the testing branch. There was a bug in how the data was decimated for plotting. Will leave this open until it's been merged into the master branch.

Nice work Mark ! 
Tnx.

closed 2023-10-26 by darksidelemm