projecthorus/radiosonde_auto_rx #816: Older (weideband) iMet decoder (Nice to have)

zanco, 2023-09-28:
About 5 times a week the older iMet's fly from different locations in the EU, none of them show on the sondehub page .

Is it possible to add support for those iMet's too ? They decode with the "regular" iMet decoder however they hava 80 kHz or a bit more bandwidth

[image]
Here you see one on 404.500 which is "overtaken" by an RS41  signal.

Thanks,
Ben

darksidelemm, 2023-09-28:
So the latest beta has a 'wideband_sondes' setting described here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Configuration-File-Changelog#version-163-in-beta

HOWEVER - note the caveats involved:
* You will not be able to decode both regular iMet sondes at the same time, and detection of other sonde types is likely to be extremely degraded. 
* Peak detection will not work - you have to set the frequencies in the always_scan list for them to be detected at all.

darksidelemm, 2023-10-26:
Closing this as support for this (at least, the best i can do) is in 1.7.1

closed 2023-10-26 by darksidelemm