projecthorus/radiosonde_auto_rx #824: Spectrum plot when using airspy/spyserver

helioshk, 2023-10-04:
Currently, the spectrum shows the full 10 MHz bandwidth, even if only 6 MHz are used to scan.
Maybe it would make sense to have 
`            f"-f {int(frequency_start)}:{int(frequency_stop)} "
`instead of
`            f"-f {_frequency_centre} "
`
When running _ss_power_cmd in sdr_wrapper.py ?

darksidelemm, 2023-10-04:
This is a limitation in the spyserver client, especially when the spyserver does not have allow_control set (which is important for auto_rx's operation). All it can really do is provide what is provided by spyserver.

As it is, while the spectrum display does show the full SDR bandwidth, peaks are only detected in the frequency start/stop ranges defined in station.cfg.

helioshk, 2023-10-04:
Understood. But I find it tidier, if auto_rx only shows the part of the spectrum which it is actually scanning. It results in a higher resolution (more pixels per Hz), which makes it easier to distinguish between noise peaks and actual signals.

closed 2024-01-24 by darksidelemm