projecthorus/radiosonde_auto_rx #827: auto_rx

PogoplugParrot, 2023-10-23:
Using the 1.6.0 and 1.6.1 versions of auto_rx, sondes are detected but the payload ID is always invalid.  This is despite the SondeHub tracker showing sondes successfully detected by stations more than 50 miles away.

The spectrum plot shows a peak at the sonde frequency well past the threshold, so it is not a  case of the sonde not being detectable.

I suspect that the issue may be the host system, which is a Pogoplug E04.  It may not have enough memory or horsepower to do decoding properly.

I cannot upgrade to 1.7.0 at this time because the build fails on this system.

Attached is a log extract showing the issue.  Any insights will be gratefully received.

2023-10-23 13:40:41,357 INFO:Task Manager - SDR #0 has been allocated to Decoder (DFM, 403.611 MHz).
2023-10-23 13:40:44,666 INFO:Decoder (RTLSDR 0) DFM 403.611 - Using fsk_demod decoder chain.
2023-10-23 13:40:44,825 INFO:Decoder (RTLSDR 0) DFM 403.611 - Starting decoder subprocess.
2023-10-23 13:40:51,892 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
2023-10-23 13:40:53,128 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
2023-10-23 13:43:44,768 ERROR:Decoder (RTLSDR 0) DFM 403.611 - RX Timed out.
2023-10-23 13:43:45,773 INFO:Decoder (RTLSDR 0) DFM 403.611 - Closed decoder subprocess.
2023-10-23 13:43:46,856 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2023-10-23 13:43:50,163 INFO:Scanner (RTLSDR 0) - Starting Scanner Thread
2023-10-23 13:43:50,167 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:44:20,211 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:44:21,293 INFO:Flask - New Web Client connected!
2023-10-23 13:44:41,209 INFO:Scanner (RTLSDR 0) - Detected peaks on 1 frequencies (MHz): [403.61]
2023-10-23 13:44:46,220 INFO:Task Manager - Detected new DFM sonde on 403.611 MHz!
2023-10-23 13:44:46,223 INFO:Halting Scanner to decode detected radiosonde.
2023-10-23 13:44:46,228 INFO:Scanner (RTLSDR 0) - Waiting for current scan to finish...
2023-10-23 13:44:56,208 INFO:Scanner (RTLSDR 0) - Scanner Thread Closed.
2023-10-23 13:44:56,213 INFO:Task Manager - SDR #0 has been allocated to Decoder (DFM, 403.611 MHz).
2023-10-23 13:44:59,659 INFO:Decoder (RTLSDR 0) DFM 403.611 - Using fsk_demod decoder chain.
2023-10-23 13:44:59,820 INFO:Decoder (RTLSDR 0) DFM 403.611 - Starting decoder subprocess.
2023-10-23 13:45:04,195 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
2023-10-23 13:45:05,925 WARNING:Payload ID DFM-xxxxxxxx is invalid. Note: DFM sondes may take a while to get an ID.
2023-10-23 13:47:36,684 INFO:Flask - New Web Client connected!
2023-10-23 13:47:59,700 ERROR:Decoder (RTLSDR 0) DFM 403.611 - RX Timed out.
2023-10-23 13:48:00,712 INFO:Decoder (RTLSDR 0) DFM 403.611 - Closed decoder subprocess.
2023-10-23 13:48:01,862 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2023-10-23 13:48:05,225 INFO:Scanner (RTLSDR 0) - Starting Scanner Thread
2023-10-23 13:48:05,236 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:48:35,219 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:49:05,215 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:49:35,221 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2023-10-23 13:50:05,207 INFO:Scanner (RTLSDR 0) - Running frequency scan.

darksidelemm, 2023-10-23:
I can't find much information about the PogoPlug series, but my guess is it is far too underpowered to do decoding. Given you are only seeing a few of the 'payload ID is invalid' lines (there should be one line per packet received), I think your system is dropping samples.

Please try again on a more suitable system.

closed 2023-10-23 by darksidelemm

PogoplugParrot, 2023-10-28:
Upgrading the host to an HP T530 thin client solved the issue.  The Pogoplug E04 does not have enough horsepower to both scan and decode in a timely fashion.  I suspect that the later Pogoplugs would also have the same problem.  Perhaps a list of host systems that are known to not work could be added to the distribution at some future time.

PogoplugParrot, 2023-10-28:
(Also posted to closed thread on topic)

An HP T530 thin client solved the issue.

darksidelemm, 2023-10-28:
Unfortunately a list of systems which are known to *not* work would be extremely long. I do say in the guide that something like a RPi 3 or newer is required.