projecthorus/radiosonde_auto_rx #832: install new

Store9w2bws, 2023-11-01:
hi i am having problems installing the latest one. like this but failed

Nov 01 23:25:24 starlink systemd[1]: Stopped auto_rx.service - auto_rx.
Nov 01 23:25:24 starlink systemd[1]: Started auto_rx.service - auto_rx.
Nov 01 23:25:25 starlink auto_rx[826]: Traceback (most recent call last):
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/core/__init__.py", line 24, in < modules>
Nov 01 23:25:25 starlink auto_rx[826]: from . import multiarray
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/core/multiarray.py", line 10, in < modules>
Nov 01 23:25:25 starlink auto_rx[826]: from . import overrides
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/core/overrides.py", line 8, in < modules>
Nov 01 23:25:25 starlink auto_rx[826]: from numpy.core._multiarray_umath import (
Nov 01 23:25:25 starlink auto_rx[826]: ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory
Nov 01 23:25:25 starlink auto_rx[826]: During handling of the above exception, another exception occurred:
Nov 01 23:25:25 starlink auto_rx[826]: Traceback (most recent call last):
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/__init__.py", line 130, in <module>
Nov 01 23:25:25 starlink auto_rx[826]: from numpy.__config__ import show as show_config
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
Nov 01 23:25:25 starlink auto_rx[826]: from numpy.core._multiarray_umath import (
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/core/__init__.py", line 50, in < modules>
Nov 01 23:25:25 starlink auto_rx[826]: raise ImportError(msg)
Nov 01 23:25:25 starlink auto_rx[826]: ImportError:
Nov 01 23:25:25 starlink auto_rx[826]: IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Nov 01 23:25:25 starlink auto_rx[826]: Importing the numpy C-extensions failed. This error can happen for
Nov 01 23:25:25 starlink auto_rx[826]: many reasons, often due to issues with your setup or how NumPy was
Nov 01 23:25:25 starlink auto_rx[826]: installed.
Nov 01 23:25:25 starlink auto_rx[826]: We have compiled some common reasons and troubleshooting tips at:
Nov 01 23:25:25 starlink auto_rx[826]: https://numpy.org/devdocs/user/troubleshooting-importerror.html
Nov 01 23:25:25 starlink auto_rx[826]: Please note and check the following:
Nov 01 23:25:25 starlink auto_rx[826]: * The Python version is: Python3.11 from "/home/pi/radiosonde_auto_rx/auto_rx/venv/bin/python3"
Nov 01 23:25:25 starlink auto_rx[826]: * The NumPy version is: "1.26.1"
Nov 01 23:25:25 starlink auto_rx[826]: and make sure that they are the versions you expect.
Nov 01 23:25:25 starlink auto_rx[826]: Please carefully study the documentation linked above for further help.
Nov 01 23:25:25 starlink auto_rx[826]: Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory
Nov 01 23:25:25 starlink auto_rx[826]: The above exception was the direct cause of the following exception:
Nov 01 23:25:25 starlink auto_rx[826]: Traceback (most recent call last):
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/auto_rx.py", line 37, in <module>
Nov 01 23:25:25 starlink auto_rx[826]: from autorx.scan import SondeScanner
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/autorx/scan.py", line 11, in <module>
Nov 01 23:25:25 starlink auto_rx[826]: import numpy as np
Nov 01 23:25:25 starlink auto_rx[826]: File "/home/pi/radiosonde_auto_rx/auto_rx/venv/lib/python3.11/site-packages/numpy/__init__.py", line 135, in <module>
Nov 01 23:25:25 starlink auto_rx[826]: raise ImportError(msg) from e
Nov 01 23:25:25 starlink auto_rx[826]: ImportError: Error importing numpy: you should not try to import numpy from
Nov 01 23:25:25 starlink auto_rx[826]: its source directory; please exit the numpy source tree, and relaunch
Nov 01 23:25:25 starlink auto_rx[826]: your python interpreter from there.
Nov 01 23:25:25 starlink systemd[1]: auto_rx.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 23:25:25 starlink systemd[1]: auto_rx.service: Failed with result 'exit-code'.

darksidelemm, 2023-11-01:
Can you try running:
```
sudo apt-get install libopenblas-dev
```
and see is that resolves the issue?

Store9w2bws, 2023-11-03:
my problem is solved. installation command overlapping problem about adding -> venv

The following commands need to be run within the ~/radiosonde_auto_rx/auto_rx directory:

$ cd ~/radiosonde_auto_rx/auto_rx/
$ python3 -m venv venv
$ source venv/bin/activate
(venv) $ pip install -r requirements.txt
A few warnings may be shown while pip does it's thing. The packages we need should still be installed however.

From now on, before attempting to run any auto_rx Python scripts, ensure you are operating within the Virtual Environment we just created. You can confirm this by running:

< key in command again > make failed to me.

$ cd ~/radiosonde_auto_rx/auto_rx/
$ source venv/bin/activate
Your terminal should have a (venv) prefix, which indicates you are operating within the venv.