projecthorus/radiosonde_auto_rx #835: Handle azimuth-only rotators better.

darksidelemm, 2023-11-22:
Add an option into the config to define that a rotator is azimuth-only. If this is enabled, only send azimuth data.

vg1320, 2023-11-23:
I have forgotten to mention that Rémi's F1NNI rotor is a Spid Rak model.

darksidelemm, 2023-11-23:
Issues with specific rotators is something you'll have to take up with hamlib. 

My intention for this fix is:
- Add a configuration parameter in the [rotator section](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L416) called 'azimuth_only', set to False by default.
- If the azimuth_only setting is set to True, then:
  - Ignore any elevation information provided by the rotctld daemon.
  - Set elevation to 0 when sending commands to rotctld.

As I don't have any azimuth-only rotator to test with, hopefully this will be sufficient. Unfortunately the hamlib 'dummy rotator' emulates an az-el rotator.

What would be useful to confirm is:
- What is the response from rotctld to a 'p' command? e.g. my az/el rotator returns two lines, one with azimuth, one with elevation. Is this the case for Rémi's rotator?

vg1320, 2023-11-23:
Yes it is the cas for Rémi's rotator :
[image]

darksidelemm, 2024-08-31:
There's a new azimuth_only option in the testing branch, v1.7.5-beta4. I've only done some very basic testing of this on my az/el rotator, so I cant really be sure if it will work correctly for an azimuth-only rotator.
If you can test this change, that would be much appreciated!

vg1320, 2024-08-31:
Thanks to Rémi F1NNI, a test of  v1.7.5-beta4 was successful this afternoon by chasing the 306-2-00201 released from Guipavas

closed 2025-03-08 by darksidelemm