projecthorus/radiosonde_auto_rx #838: Update APRS Serve

michaelself10, 2023-12-12:
I am trying to get my receiving data uploaded to radiosondy.info but I believe the port number has changed. On the website it shows s1.radiosondy.info:14580 or radiosondy.info:14580 instead of the automatic 14590 that the program reverts to. Is there a way for me to change this?

darksidelemm, 2023-12-12:
Port 14590 is the 'non-forwarding' port, which doesn't forward data out to the wider APRS-IS network. Of course, neither port 14580 or 14590 forward to APRS-IS anymore, but port 14590 still exists. 

On the latest version of auto_rx both port numbers 14580 and 14590 should be allowed. The server still has to be 'radiosondy.info', but this shouldn't be an issue?

Please re-open this if you can't use port 14580 on the latest version of auto_rx.

closed 2023-12-12 by darksidelemm

michaelself10, 2023-12-13:
> Port 14590 is the 'non-forwarding' port, which doesn't forward data out to the wider APRS-IS network. Of course, neither port 14580 or 14590 forward to APRS-IS anymore, but port 14590 still exists. [...]

I updated the software and it does now allow me to use port 14580, but it is still not uploading to radiosondy.info

darksidelemm, 2023-12-16:
What are you seeing in your logs? Is there an error of some sort? 
Are you seeing log messages indicating uploads to APRS-IS?