projecthorus/radiosonde_auto_rx #839: Cannot receive RS41 sonde.

LaboratorioGluon, 2023-12-14:
I am trying to receive a RS41 sonde in 403.000 MHz (from Madrid, Spain)

I have attached the same antenna I am using, with an RTL-SDR and check that I am receiving it.
Also, when I look in the spectrogram screen, I see a peak near 403.000 Mhz.

However, there is no info from the sonde, and nothing is received. I have connected to the serial and I see a message saying "TOO MUCH DATA" every now and then... (see log at the end)

Do you know what could be wrong? 

Thanks in advance

```MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 127376, unused stack: 5096
Exact bitrate is 4799.98
Sonde::setup() done: Type RS41 Freq 403.001952, AFC BW: 12500, RX BW: 6250
RSSI update: 0/2
TOO MUCH DATA
Sonde::timeoutEvent: NORX
Sonde:receive(): result 1 (RX_TIMEOUT), event 0b => action 82
RXtask: start DECODER for sonde 2 (was DECODER)
Sonde::setup() start on index 2
waitRXcomplete returning 8201 (RX_TIMEOUT)
MAIN: loopDecoder: action 82 (Sonde=2) => 65  [current: main=2, rxtask=2]
MAIN: updateDisplay started
MAIN: updateDisplay done (after 67 ms)

MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 127376, unused stack: 5096
Exact bitrate is 4799.98
Sonde::setup() done: Type RS41 Freq 403.001952, AFC BW: 12500, RX BW: 6250
Sonde::timeoutEvent: NORX
Sonde:receive(): result 1 (RX_TIMEOUT), event 0b => action 82
RXtask: start DECODER for sonde 2 (was DECODER)
Sonde::setup() start on index 2
waitRXcomplete returning 8201 (RX_TIMEOUT)
MAIN: loopDecoder: action 82 (Sonde=2) => 65  [current: main=2, rxtask=2]
MAIN: updateDisplay started
MAIN: updateDisplay done (after 67 ms)

MAIN: Running loop in state 0 [currentDisp:0, lastDisp:0]. free heap: 127376, unused stack: 5096
Exact bitrate is 4799.98
Sonde::setup() done: Type RS41 Freq 403.001952, AFC BW: 12500, RX BW: 6250
Sonde::timeoutEvent: NORX
Sonde:receive(): result 1 (RX_TIMEOUT), event 0b => action 82
waitRXcomplete returning 8201 (RX_TIMEOUT)
MAIN: loopDecoder: action 82 (Sonde=2) => 65  [current: main=2, rxtask=2]
MAIN: updateDisplay started
RXtask: start DECODER for sonde 2 (was DECODER)
Sonde::setup() start on index 2
MAIN: updateDisplay done (after 70 ms)

....
``

darksidelemm, 2023-12-14:
I don't know what software you are running, but that output does not look to be from radiosonde_auto_rx.

LaboratorioGluon, 2023-12-15:
Sorry my bad! I missclicked the link

closed 2023-12-15 by LaboratorioGluon