projecthorus/radiosonde_auto_rx #840: rtl_power call failed with return code -2

michaelself10, 2023-12-30:
I want to make sure this error code is nothing serious, or anything at all. Sometimes when I quit the code (ctrl+c), I get the following error message...
[image]

darksidelemm, 2023-12-30:
That's fine - this is because the ctrl-c has probably killed that rtl_power subprocess, resulting in an abnormal return code. Feel free to close the issue if that has answered your questions.

michaelself10, 2023-12-30:
> That's fine - this is because the ctrl-c has probably killed that rtl_power subprocess, resulting in an abnormal return code. Feel free to close the issue if that has answered your questions.

Thanks! Just making sure it is all good, I will close the issue now.

closed 2023-12-30 by michaelself10