projecthorus/radiosonde_auto_rx #841: feature request. allow for frequency time out in Advanced Controls menu

lelun, 2023-12-31:
as stated requesting that a frequency time out button is added to the advanced controls menu. main idea is that if that button is used it will stop decoding and time out that frequency for a predetermined amount of time. 

i know there is a automatic feature that times out frequencies but some times its useful to stop decoding and restart it. so i dont have that enabled due to how rearly its usefull

but some rear times its useful to stop decoding of a certain sonde (like if its a fualty unit) and get autorx ready to recive any follow up sondes. its not a critical feature but a nice to have for those rear occasions the automatic launcher mangles one on the way out. 

it would be nice to have the option to do a temperary time out. without going in to the settings to configure it for a single sonde. 

also merry Christmas happy New year. and good luck with your hunts in to the new year :)

darksidelemm, 2023-12-31:
How about a second button near the 'stop decoder' button called 'Temp Lockout', which stops and locks out the selected frequency? 
The lockout time would still be based off what's set in the temporary_block_time setting in station.cfg - https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L546

darksidelemm, 2024-01-01:
I've added this function in the testing branch:
[image]

lelun, 2024-01-01:
> I've added this function in the testing branch: <img alt="Screen Shot 2024-01-01 at 10 32 36" width="433" src="https://private-user-images.githubusercontent.com/1040275/293525750-f383abb9-fc0a-4df7-96b2-a3fc2e290e3c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQwNzE0NjIsIm5iZiI6MTcwNDA3MTE2MiwicGF0aCI6Ii8xMDQwMjc1LzI5MzUyNTc1MC1mMzgzYWJiOS1mYzBhLTRkZjctOTZiMi1hM2ZjMmUyOTBlM2MucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDFUMDEwNjAyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWFkYzM4ODNhYzM1ZWY3OWFmYjdmMDllMDZmYTZiOWZkZWJjMjUzYTE2Y2RkZjc1OTE4ZmQ0OTVmMmJlZThhYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9u3TORaqLhwaPztuczONjgA9KG4xcuDCbwYCvGkbJEM">

yes somthing like this would be perfect.

lelun, 2024-01-01:
tested and it works. Thanks for this feature happy new year :)

closed 2024-01-01 by lelun