projecthorus/radiosonde_auto_rx #843: Coverage calculation - how does it work?

MikeTango, 2024-01-01:
Hi, 

i am looking at the plots for my home setup and noticed that entire paths of sondes are not in the "coverage" polygon despite being completely logged. Is there any way to fix that?

regards,

MT
[image]

darksidelemm, 2024-01-01:
The coverage polygon only uses first and last positions, not every single position. This is to avoid having to read and parse every entry of what might be a very large number of log files.

MikeTango, 2024-01-02:
Ok, that makes a bit of sense because mostly the balloons don't boomerang, but leaves these odd gaps. How do the single lines and very thin spikes come about? Shouldn't the polygon connect the extremes?

closed 2024-11-16 by darksidelemm