projecthorus/radiosonde_auto_rx #844: frequency correction?

radiohound, 2024-01-05:
When my SDR hears a weather balloon on 404.000 mhz, it thinks the weather balloon is on 403.980 mhz. The signal is heard fine, its just an issue of reporting the wrong frequency. Is there a way in the software to correct for this reporting error?

Thank you,

Walter

darksidelemm, 2024-01-05:
If you are using a RTLSDR, the station.cfg file has an option for a ppm offset setting: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L71

However, I'm wondering what SDR you are using that results in a 20 kHz frequency offset. I've found that most decent quality RTLSDRs (e.g. the RTLSDR v3 ones) are pretty much on-frequency from the factory.

radiohound, 2024-01-07:
Thank you very much for your help. I was able to make that change to the config file. I am using an old DVB-T-DAB, only because I already had it. It's somewhat surprising that I was getting what I thought was pretty decent reception with it, although it's pretty far off frequency. I have purchased a RTL SDR V3 which I will get shortly, and will probably perform even better.

closed 2024-01-07 by radiohound