projecthorus/radiosonde_auto_rx #847: container crash : Main Loop Error - 1.7.1.1 is not valid SemVer string

heneault, 2024-01-06:
Latest commit https://github.com/projecthorus/radiosonde_auto_rx/commit/15e0a7010cca837a7fb69ccdc5e0657ae3c19a84#

seems to break latest container version

Traceback (most recent call last):
 main    File "/opt/auto_rx/auto_rx.py", line 1103, in <module>
 main      main()
 main    File "/opt/auto_rx/auto_rx.py", line 1066, in main
 main      version_startup_check()
 main    File "/opt/auto_rx/autorx/utils.py", line 147, in version_startup_check
 main      _newer_version = check_autorx_versions()
 main    File "/opt/auto_rx/autorx/utils.py", line 136, in check_autorx_versions
 main      if semver.compare(_main_branch_version, current_version) == 1:
 main    File "/root/.local/lib/python3.9/site-packages/semver/_deprecated.py", line 80, in wrapper
 main      return func(*args, **kwargs)  # type: ignore
 main    File "/root/.local/lib/python3.9/site-packages/semver/_deprecated.py", line 112, in compare
 main      return Version.parse(ver1).compare(ver2)
 main    File "/root/.local/lib/python3.9/site-packages/semver/version.py", line 646, in parse
 main      raise ValueError(f"{version} is not valid SemVer string")
 main  Main Loop Error - 1.7.1.1 is not valid SemVer string

darksidelemm, 2024-01-06:
Oh bugger. OK, I'll have to bump the version to 1.7.2, what a pain.

heneault, 2024-01-06:
would be nice to add an option in the confi to disable the version check

darksidelemm, 2024-01-06:
I'd rather not do that. I want users to know there is a new version available.

I've bumped the master version to 1.7.2, and testing to 1.7.3-beta4.

Builds are in progress: https://github.com/projecthorus/radiosonde_auto_rx/actions

closed 2024-01-06 by darksidelemm