projecthorus/radiosonde_auto_rx #853: SondeHub Upload Stopping

michaelselfwx, 2024-01-24:
So recently I have been having a problem with uploading the radiosonde telemetry to SondeHub. I am using a Rpi 4 with 4 rtl-sdr receivers attached to one 70cm Ham beam antenna (located in Gainesville, FL, USA). Recently, it has uploaded the telemetry of the Jacksonville station's radiosonde (about 60km from receiver) to SondeHub but only for about 20-30 minutes. I can see on the web interface that it receives the radiosonde for much longer after that as well as it is still uploading to radiosondy.info. I have looked at the program while it is running and there are no critical error codes about SondeHub. Anyone else have this problem before?

darksidelemm, 2024-01-24:
I'm going to need more information to work out what's going on:
- Do you have a few specific radiosonde serial numbers where this issue is occuring?
- Please set the `save_system_log` setting ([here](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L453) to `True`, and restart auto_rx. This will make it easier for you to send me your auto_rx system log file.
- Wait for the issue to happen again, and then send me the log file (vk5qi@rfhead.net)

darksidelemm, 2024-11-16:
Without any further information, not much I can do here. If you can replicate this on the latest version and get me logs, then feel free to re-open the issue.

closed 2024-11-16 by darksidelemm