projecthorus/radiosonde_auto_rx #855: log Parameter not working for rtl_power

skymaker-c2is, 2024-01-26:
Hi,

I'm trying to log the decoder outputs to /var/tmp with

`auto_rx.py -c /etc/default/radiosondeautorx -l /var/tmp/radiosondeautorx -t 0`

Unfortunately, the output of the scanner (rtl_power) is not logged to the directory given with -l. auto_rx is trying to log the output to the working directory, which isn't writeable for the user running auto_rx:

```
Jan 27 00:29:40 futro-db auto_rx[126773]: 2024-01-27 00:29:40,906 DEBUG:Scanner (RTLSDR 00001001) - Running command: timeout -k 30  30  rtl_power -p 0 -d 00001001 -f 400050000.0:406000000.0>
Jan 27 00:29:41 futro-db auto_rx[126773]: 2024-01-27 00:29:41,464 CRITICAL:Scanner (RTLSDR 00001001) - rtl_power call failed with return code 1.
Jan 27 00:29:41 futro-db auto_rx[126773]: 2024-01-27 00:29:41,464 CRITICAL:Scanner (RTLSDR 00001001) - rtl_power reported error: Found 3 device(s):
Jan 27 00:29:41 futro-db auto_rx[126773]:   0:  Realtek, RTL2838UHIDIR, SN: 10000001
Jan 27 00:29:41 futro-db auto_rx[126773]:   1:  Realtek, RTL2838UHIDIR, SN: 00001001
Jan 27 00:29:41 futro-db auto_rx[126773]:   2:  Realtek, RTL2838UHIDIR, SN: 10000010
Jan 27 00:29:41 futro-db auto_rx[126773]: Using device 1: Generic RTL2832U OEM
Jan 27 00:29:41 futro-db auto_rx[126773]: Number of frequency hops: 3
Jan 27 00:29:41 futro-db auto_rx[126773]: Dongle bandwidth: 2644444Hz
Jan 27 00:29:41 futro-db auto_rx[126773]: Downsampling by: 1x
Jan 27 00:29:41 futro-db auto_rx[126773]: Cropping by: 25.00%
Jan 27 00:29:41 futro-db auto_rx[126773]: Total FFT bins: 12288
Jan 27 00:29:41 futro-db auto_rx[126773]: Logged FFT bins: 9216
Jan 27 00:29:41 futro-db auto_rx[126773]: FFT bin size: 645.62Hz
Jan 27 00:29:41 futro-db auto_rx[126773]: Buffer size: 16384 bytes (3.10ms)
Jan 27 00:29:41 futro-db auto_rx[126773]: Reporting every 20 seconds
Jan 27 00:29:41 futro-db auto_rx[126773]: Found Rafael Micro R820T tuner
Jan 27 00:29:41 futro-db auto_rx[126773]: Tuner gain set to automatic.
Jan 27 00:29:41 futro-db auto_rx[126773]: Failed to open log_power_00001001.csv
```

When the user has write access to the working directory, everything works fine.
I don't know if this is intended, but it would be nice if auto_rx would either log the output of rtl_power to the directory given by -l or to another directory given by another parameter.

darksidelemm, 2024-01-26:
Yes, right now auto_rx expects the working directory to be writable. The rtl_power outputs are only considered temporary files, so I didn't want to put them into the log directory. 
However, moving them to the log directory looks to be fairly trivial, so I'll look at doing that in the testing branch soon.

darksidelemm, 2024-01-27:
Ok, this was pretty simple. I've made the change in the testing branch.

I haven't yet looked to see if there's any other files auto_rx saves into the working directory.

skymaker-c2is, 2024-01-28:
Thank you for the quick fix! I patched the stable release locally and it's working.

closed 2024-01-28 by skymaker-c2is