projecthorus/radiosonde_auto_rx #872: tracker.sondehub.org auto-zoom on click?

K4KDR, 2024-05-02:
Not an autoRX issue specifically, but a quick question / request in regard to **tracker.sondehub.org**

When a balloon is clicked on, the map automatically zooms in several steps.  So, after every signal click (to see the track plot of that particular balloon) you have to zoom back out & re-center the map to show the geographic are that is of interest to you.

Would you consider - or is it even possible - to make this behavior optional by way of an on/off switch on the **Settings** tab?

Many thanks!

-Scott,  K4KDR

darksidelemm, 2024-05-02:
If this is a sondehub tracker issue, then please raise the issue on that repository: https://github.com/projecthorus/sondehub-tracker

closed 2024-05-02 by darksidelemm

K4KDR, 2024-05-02:
Thanks a million!   Was not aware the the website had a github repository.   Much appreciated.

For anyone who finds this issue from a search, my question was addressed previously at:

https://github.com/projecthorus/sondehub-tracker/issues/240