projecthorus/radiosonde_auto_rx #874: MRZ-N1 does not decode over SpyServer

DJRaov, 2024-05-13:
Hi!

For the past 4 days I have been trying my best at decoding the (supposedly) MRZ-N1 sondes that have been passing above me lately, with no luck. Even though the signal is definitely strong enough I get no frames, not with the "regular" decode chain, not with the experimental/fsk_demod chain. I tried decoding it manually, and everything I found about the N1 matches what I am getting. What could I be missing here?

A signal sample is available if required: [https://ub8qbd.satdump.org/files/mrz-n1_raw.wav](https://ub8qbd.satdump.org/files/mrz-n1_raw.wav)

darksidelemm, 2024-05-13:
Yep, I see that detects as a MRZ, but I'm not sure it is one.
```
% cat ~/Downloads/mrz-n1_raw.wav| ./dft_detect --iq --bw 20 --dc - 48000 16
MRZ: 0.9831 , +632.1Hz
```

Passing that signal through the MRZ decode chain just results in gibberish data:
```
cat ~/Downloads/mrz-n1_raw.wav| ./fsk_demod --cs16 -s -b -23000 -u 23000 2 48000 2400 - - | ./mp3h1mod --auto --json --softin --ptu
Setting estimator limits to -23000 to 23000 Hz.
 [ 7]  (11:34:40)  lat: -18.05036  lon: -163.20724  alt: 13154986.29   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 8]  (11:34:42)  lat: -29.86333  lon: -36.86229  alt: 22635902.24   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 8]  (11:34:42)  lat: -29.86333  lon: -36.86229  alt: 22635902.24   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 2]  (02:200:138)  lat: 46.34215  lon: -145.82055  alt: 3451335.02   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 2]  (66:200:139)  lat: 40.13625  lon: -127.46280  alt: 1008367.23   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 4]  (44:136:176)  lat: -28.39585  lon: 131.67823  alt: 14825653.78   vH:  0.0  D:   0.0  vV: 0.0   [NO]
 [ 2]  (200:139:04)  lat: 30.35747  lon: 29.52724  alt: 15273396.40   vH:  0.0  D:   0.0  vV: 0.0   [NO]
...
```

@rs1729 Might have some idea of what this might be. 

On a related note, i think the mp3h1mod arguments on this line (the 'non-experimental' decoder) might be wrong:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/decode.py#L614

As I get:
```
% cat ~/Downloads/mrz-n1_raw.wav| ./mp3h1mod --IQ 0.0 --lp - 48000 16 --json --ptu
error open --lp
```

rs1729, 2024-05-13:
Can you tell us the approximate location?
Is the time 11:34:40 (UTC/GPS) correct?

DJRaov, 2024-05-13:
The sondes are launched from Kurgan, Russia, and the time is indeed correct.

rs1729, 2024-05-13:
So position N55.44463  E65.37257 is also right.
OK.
I think it is
```
 [ 7]  (11:34:40)  lat: 55.44463  lon: 65.37257  alt: 1195.20  [OK]
 [ 8]  (11:34:42)  lat: 55.44481  lon: 65.37254  alt: 1202.30  [OK]
 [ 9]  (11:34:44)  lat: 55.44501  lon: 65.37233  alt: 1209.80  [OK]
 [10]  (11:34:46)  lat: 55.44528  lon: 65.37226  alt: 1218.50  [OK]
 [11]  (11:34:48)  lat: 55.44546  lon: 65.37223  alt: 1227.50  [OK]
 [12]  (11:34:50)  lat: 55.44569  lon: 65.37204  alt: 1235.10  [OK]
```
So it is probably a different radiosonde similar to the MRZ we know, older or newer. Don't really know much about these radiosondes.

Looks like the same header. Would be good to have something to tell them apart.
The frames are a bit shorter.

Could you make a longer recording, 1 minute or 2?

DJRaov, 2024-05-13:
> So it is probably a different radiosonde similar to the MRZ we know, older or newer. Don't really know much about these radiosondes.
I looked around the Radiy website, no models other than the N1 matched the RF signal. Maybe new board revision?

> Could you make a longer recording, 1 minute or 2?
I got a recording right as the sonde got released: https://ub8qbd.satdump.org/files/mrz-n1_raw_long.wav

rs1729, 2024-05-13:
Temperature looks ok, humidity could be ok, although not exactly what the soundings report.
http://weather.uwyo.edu/cgi-bin/sounding?region=mideast&TYPE=TEXT%3ALIST&YEAR=2024&MONTH=05&FROM=1312&TO=1312&STNM=28661

```
 [ 2]  (11:30:46)  lat: 55.45423  lon: 65.39791  alt: 189.50   vV: 4.5   [OK]
 [ 3]  (11:30:48)  lat: 55.45398  lon: 65.39777  alt: 199.40   vV: 4.5   [OK]
 [ 4]  (11:30:50)  lat: 55.45379  lon: 65.39753  alt: 208.40   vV: 5.1   [OK]
 [ 5]  (11:30:52)  lat: 55.45349  lon: 65.39732  alt: 217.90   vV: 3.9   [OK]
 [ 6]  (11:30:54)  lat: 55.45325  lon: 65.39716  alt: 228.30   vV: 3.9   [OK]
 [ 7]  (11:30:56)  lat: 55.45300  lon: 65.39693  alt: 238.10   vV: 3.9   [OK]
 [ 8]  (11:30:58)  lat: 55.45274  lon: 65.39680  alt: 247.40   vV: 3.9   [OK]
 [ 9]  (11:31:00)  lat: 55.45252  lon: 65.39663  alt: 256.60   vV: 3.9   [OK]
 [10]  (11:31:02)  lat: 55.45226  lon: 65.39645  alt: 263.90   vV: 3.9   [OK]
 [11]  (11:31:04)  lat: 55.45206  lon: 65.39632  alt: 272.10   vV: 3.9   [OK]
 [12]  (11:31:06)  lat: 55.45177  lon: 65.39614  alt: 278.70   vV: 3.9   [OK]  <13> snC: 401113
 [13]  (11:31:08)  lat: 55.45158  lon: 65.39602  alt: 287.80   vV: 3.9   [OK]  <14> snD: 406285
 [14]  (11:31:10)  lat: 55.45132  lon: 65.39575  alt: 297.20   vV: 4.5   [OK]  <15> calDate: 180424
 [15]  (11:31:12)  lat: 55.45109  lon: 65.39563  alt: 306.40   vV: 4.5   [OK]  <16> 2024-05-13
 [ 0]  (11:31:14)  lat: 55.45088  lon: 65.39534  alt: 317.10   vV: 5.2   [OK]
 [ 1]  (11:31:16)  lat: 55.45057  lon: 65.39518  alt: 324.00   vV: 4.5   T=2.8C RH=98%   [OK]
 [ 2]  (11:31:18)  lat: 55.45039  lon: 65.39496  alt: 330.50   vV: 3.9   T=2.7C RH=99%   [OK]
 [ 3]  (11:31:20)  lat: 55.45012  lon: 65.39469  alt: 337.90   vV: 4.5   T=2.7C RH=99%   [OK]
 [ 4]  (11:31:22)  lat: 55.44994  lon: 65.39453  alt: 345.00   vV: 4.5   T=2.6C RH=99%   [OK]
 [ 5]  (11:31:24)  lat: 55.44968  lon: 65.39424  alt: 353.30   vV: 4.5   T=2.6C RH=99%   [OK]
 [ 6]  (11:31:26)  lat: 55.44946  lon: 65.39407  alt: 362.60   vV: 3.9   T=2.5C RH=99%   [OK]
 [ 7]  (11:31:28)  lat: 55.44922  lon: 65.39376  alt: 372.90   vV: 5.2   T=2.5C RH=99%   [OK]
 [ 8]  (11:31:30)  lat: 55.44899  lon: 65.39361  alt: 382.20   vV: 3.9   T=2.4C RH=100%   [OK]
 [ 9]  (11:31:32)  lat: 55.44880  lon: 65.39335  alt: 390.10   vV: 4.5   T=2.4C RH=100%   [OK]
 [10]  (11:31:34)  lat: 55.44858  lon: 65.39310  alt: 397.00   vV: 4.5   T=2.3C RH=100%   [OK]
 [11]  (11:31:36)  lat: 55.44836  lon: 65.39292  alt: 402.70   vV: 3.9   T=2.3C RH=100%   [OK]
 [12]  (11:31:38)  lat: 55.44813  lon: 65.39262  alt: 408.70   vV: 3.9   T=2.2C RH=100%   [OK]  <13> snC: 401113
 [13]  (11:31:40)  lat: 55.44785  lon: 65.39249  alt: 416.20   vV: 3.9   T=2.2C RH=100%   [OK]  <14> snD: 406285
 [14]  (11:31:42)  lat: 55.44762  lon: 65.39216  alt: 424.00   vV: 4.5   T=2.1C RH=100%   [OK]  <15> calDate: 180424
 [15]  (11:31:44)  lat: 55.44737  lon: 65.39194  alt: 432.60   vV: 4.5   T=2.1C RH=100%   [OK]  <16> 2024-05-13
 [ 0]  (11:31:46)  lat: 55.44720  lon: 65.39165  alt: 440.30   vV: 5.2   T=2.0C RH=100%   [OK]
 [ 1]  (11:31:48)  lat: 55.44692  lon: 65.39136  alt: 447.20   vV: 3.9   T=2.0C RH=100%   [OK]
 [ 2]  (11:31:50)  lat: 55.44674  lon: 65.39115  alt: 456.40   vV: 3.9   T=1.9C RH=100%   [OK]
 [ 3]  (11:31:52)  lat: 55.44650  lon: 65.39083  alt: 465.30   vV: 3.9   T=1.9C RH=100%   [OK]
 [ 4]  (11:31:54)  lat: 55.44626  lon: 65.39060  alt: 473.50   vV: 3.9   T=1.8C RH=100%   [OK]
 [ 5]  (11:31:56)  lat: 55.44608  lon: 65.39032  alt: 482.70   vV: 3.9   T=1.8C RH=100%   [OK]
 [ 6]  (11:31:58)  lat: 55.44586  lon: 65.39007  alt: 492.10   vV: 3.9   T=1.7C RH=100%   [OK]
 [ 7]  (11:32:00)  lat: 55.44572  lon: 65.38985  alt: 498.50   vV: 3.9   T=1.7C RH=100%   [OK]
 [ 8]  (11:32:02)  lat: 55.44551  lon: 65.38961  alt: 504.60   vV: 3.9   T=1.6C RH=100%   [OK]
 [ 9]  (11:32:04)  lat: 55.44532  lon: 65.38941  alt: 510.90   vV: 3.9   T=1.6C RH=100%   [OK]
 [10]  (11:32:06)  lat: 55.44512  lon: 65.38912  alt: 517.90   vV: 3.9   T=1.6C RH=100%   [OK]
 [11]  (11:32:08)  lat: 55.44488  lon: 65.38892  alt: 525.40   vV: 4.5   T=1.5C RH=100%   [OK]
 [12]  (11:32:10)  lat: 55.44467  lon: 65.38856  alt: 534.00   vV: 4.5   T=1.5C RH=100%   [OK]  <13> snC: 401113
 [13]  (11:32:12)  lat: 55.44444  lon: 65.38835  alt: 543.70   vV: 3.9   T=1.4C RH=100%   [OK]  <14> snD: 406285
 [14]  (11:32:14)  lat: 55.44427  lon: 65.38810  alt: 554.50   vV: 4.5   T=1.4C RH=100%   [OK]  <15> calDate: 180424
 [15]  (11:32:16)  lat: 55.44402  lon: 65.38779  alt: 564.50   vV: 5.1   T=1.3C RH=100%   [OK]  <16> 2024-05-13
 [ 0]  (11:32:18)  lat: 55.44378  lon: 65.38757  alt: 575.60   vV: 3.9   T=1.3C RH=100%   [OK]
 [ 1]  (11:32:21)  lat: 55.44347  lon: 65.38705  alt: 590.20   vV: 5.8   T=1.2C RH=100%   [OK]
 [ 2]  (11:32:22)  lat: 55.44334  lon: 65.38695  alt: 594.20   vV: 4.5   T=1.2C RH=100%   [OK]
 [ 3]  (11:32:24)  lat: 55.44316  lon: 65.38667  alt: 602.80   vV: 5.1   T=1.2C RH=100%   [OK]
 [ 4]  (11:32:26)  lat: 55.44295  lon: 65.38627  alt: 611.00   vV: 5.2   T=1.2C RH=100%   [OK]
 [ 5]  (11:32:28)  lat: 55.44275  lon: 65.38604  alt: 620.30   vV: 4.5   T=1.1C RH=100%   [OK]
 [ 6]  (11:32:30)  lat: 55.44257  lon: 65.38567  alt: 628.10   vV: 5.2   T=1.1C RH=100%   [OK]
 [ 7]  (11:32:32)  lat: 55.44232  lon: 65.38540  alt: 635.00   vV: 4.5   T=1.1C RH=100%   [OK]
 [ 8]  (11:32:34)  lat: 55.44217  lon: 65.38504  alt: 642.20   vV: 4.5   T=1.0C RH=100%   [OK]
[... cut ...]
Not sure about velocity, maybe more recordings could clear the picture, after burst e.g.
Config and (P)TU data seems to be the same.
Recording of higher altitudes with lower humidity could also help.

To tell them apart, the different position of the FFFF dividing position and temperature/humidity could the way to go.

I think it should be possible to include this in the mp3h1mod-decoder. If we knew what the difference to the other type is, we could put a subtype or new type in the JSON output.

Maybe I can make a test version if you want to observe and maybe try to recover one.

DJRaov, 2024-05-14:
> Maybe I can make a test version if you want to observe and maybe try to recover one.

That would be great, as it seems like no other software can decode the revision that is being launched near me. 
I'll also record another launch bit later today and send it once I lose the signal.

DJRaov, 2024-05-14:
Well, that took me a bit. Got lucky with the wind today - sonde flew closer to the antenna and didn't have as much fading as compared to last 2 passes: [https://ub8qbd.satdump.org/files/mrz-n1_pass2.wav](https://ub8qbd.satdump.org/files/mrz-n1_pass2.wav)

rs1729, 2024-05-15:
ok,
here is a first test version
https://github.com/rs1729/RS/blob/test/demod/mod/mrz2mod_test.c
I didn't look at the velocity yet.

You can compile it the same way as mp3h1mod.c:
```
gcc mrz2mod_test.c demod_mod.o -lm -o mrz2test
```
Should work with both, the old ecef coordinates and lat/lon for this new(?) type.

If you have recordings from/after burst, this would be interesting, too.

DJRaov, 2024-05-15:
> here is a first test version
> https://github.com/rs1729/RS/blob/test/demod/mod/mrz2mod_test.c

Thanks a lot, will get back with results after launch.

> If you have recordings from/after burst, this would be interesting, too.
Antenna is too bad for that for now (CB 5/8 monopole). I will build a 1/4 GP for 403MHz soon, hopefully it'll improve things.

DJRaov, 2024-05-15:
Well, it's getting there. Seems to decode now, but auto_rx itself does not want to accept the data:
[image]

darksidelemm, 2024-05-15:
If auto_rx sees a 'sats' field in the incoming JSON from the decoder, it'll act on it as you have found. If that field is 0, and that gets sent on to SondeHub, the the SondeHub tracker will not display that telemetry.

If we don't know where the equivalent field is in this variant, then i'd suggest not including the 'sats' field in the JSON at all, though this does risk bad data coming through. Hopefully the information can be found somewhere in the telemetry.

DJRaov, 2024-05-15:
> If that field is 0, and that gets sent on to SondeHub, the the SondeHub tracker will not display that telemetry.

According to Sondehub the sat count sent is 255, I'll have it not send the sat count after I lose signal.
[image]

> Hopefully the information can be found somewhere in the telemetry.

I recorded the current launch, hopefully it'll prove useful: [https://ub8qbd.satdump.org/files/mrz-n1_pass3.wav](https://ub8qbd.satdump.org/files/mrz-n1_pass3.wav)

rs1729, 2024-05-15:
Velocity and Sats are not included (yet). I was thinking about excluding the fields in the JSON output, was not sure, if you wanted to use it in auto_rx.
(look at the latest update)

Don't know if auto_rx excepts packets without sat > 3 or 4, I believe it checks for bad GPS signals.

Also I could not check if time is UTC or GPS, with the other GNSS chip it was UTC I believe.
@DJRaov maybe you can check, if the decoded time is UTC or 18 seconds offset (GPS).
Altitude GPS or MSL is also not clear for this type.

"type" is still "MRZ", maybe we need a subtype, when we know more.

DJRaov, 2024-05-15:
> Don't know if auto_rx excepts packets without sat > 3 or 4

4 sats minimum:
[image]

> @DJRaov maybe you can check, if the decoded time is UTC or 18 seconds offset (GPS).
> Altitude GPS or MSL is also not clear for this type.

Don't think I can reliably check time from the current launch, I'll see what I can do tomorrow.
Same for GPS/MSL altitude, I'll see if I can decode the sonde right as it is released.

rs1729, 2024-05-15:
The data on the sondehub server suggests it is still UTC:
```
        "time_received": "2024-05-15T11:45:22.442472Z",
        "datetime": "2024-05-15T11:45:20.000000Z",
```

If it would be GPS time, it would look like it was received ahead of time, e.g. RS41:
```
        "time_received": "2023-02-27T04:53:20.256757Z",
        "datetime": "2023-02-27T04:53:36.000000Z",
```
If it is a combined GPS/GLONASS receiver, maybe UTC is the better choice.
If the number of Sats is combined GPS/GLONASS, it is more difficult to find a plausible field. First I look at velocity.

With "GPS-Altitude" I mean above ellipsoid.
Altitude MSL/GPS is not so easy, either you need a launch together with a known radiosonde, or you have data from the ground of a recovered radiosonde or before launch. With ECEF coordinates it could not be MSL, I guess. But here we have Lat/Lon.

DJRaov, 2024-05-15:
>  you have data from the ground of a recovered radiosonde or before launch

That's exactly what I'm betting on, I'll be close to the site at launch.

rs1729, 2024-05-15:
Geoid Height at your area should be -20m, i.e.
100m MSL (above Geoid) would be 80m above Ellipsoid.

If you observe the launch, maybe the radiosonde is turned on inside a building/shelter where it does not have a valid position or low satellite count. However the radiosonde could be turned on half an hour before launch or even more.

rs1729, 2024-05-15:
@DJRaov 
On Sondehub I see data from burst and descent. Do you have recordings of that part? Or raw data?

DJRaov, 2024-05-15:
That's from an entirely different station, no idea where it is and what software it runs, though.

rs1729, 2024-05-15:
Ah, okay.
Horizontal velocity is just 16 bit speed and 2 bit heading. The byte after that might be Sats (GPS+GLONASS).
After that 2 times 16 bit, does not really look like vertical speed, compared to the altitude difference. Some radiosondes don't have it, just very basic NMEA output. After burst there would be change of sign.

rs1729, 2024-05-17:
Took a closer look at the data, only 3 bytes left for Sats or perhaps battery voltage.
There are two 16-bit values before the FFFF that match the calculated temperature/humidity values pretty good for the old (ECEF) type. The new type (LatLon) temperature values are also matching good, but the rel. humidity differs a bit more for lower temperatures. So I switched to the 16-bit values in `mrz2mod_test.c`. First I thought the 16-bit T could be internal temperature or from the humidity sensor. But why would it be grouped with the 16-bit RH then? Looks like after FFFF there is raw data, and for the new type the calculation is a bit different. I believe there was another document describing the calculation a bit differently, but for now the 16-bit values seem to be good enough or better than the calculations.

So it looks like there is no vertical velocity for this new type.
With the number of satellites I'm not sure, probably one of the 3 remaining values, maybe the first, if it is the sum of different GNSS systems, GPS+GLONASS perhaps. The other two bytes could also be some status or accuracy.
Observing the startup or a recovered radiosonde could give some more hints.

DJRaov, 2024-05-18:
> With the number of satellites I'm not sure, probably one of the 3 remaining values, maybe the first, if it is the sum of different GNSS systems, GPS+GLONASS perhaps. 

Current sondes do indeed use both GPS and GLONASS for positioning, though I don't exactly see why the sat count wouldn't be combined into one value.

> Observing the startup or a recovered radiosonde could give some more hints.

Did not catch it booting up, but I did get it just before the launch and as it was released: [https://ub8qbd.satdump.org/files/mrz-n1_prelaunch.wav](https://ub8qbd.satdump.org/files/mrz-n1_prelaunch.wav) and [https://ub8qbd.satdump.org/files/mrz-n1_postlaunch.wav](https://ub8qbd.satdump.org/files/mrz-n1_postlaunch.wav)
Weather also seems to be favorable for recovery of tomorrow's night launch, I'll leave the decoder running and see where it ends up.

rs1729, 2024-05-19:
Altitude at launch site is 70m.
The radiosonde shows
```
(11:19:08)  lat: 55.45667  lon: 65.40013  alt: 85.40   vH:  0.0  D: 164.9  T=19.60C RH=36.36%   [OK]  <16> 2024-05-18
```
with a high number of what is assumed to be the number of satellites.
Geoid altitude is ca. -20m, so GPS altitude would  be 50m above ellipsoid.
However the radiosonde shows rather the opposite, so it looks more like radiosonde altitude is MSL.
Just before launch the "Sats" field drops to 8, then goes up again. Would make sense, maybe it was moved before release.
```
 (11:29:46)  lat: 55.45696  lon: 65.40024  alt: 76.20   vH:  0.1  D: 293.7
 (11:30:04)  lat: 55.45683  lon: 65.40019  alt: 96.10   vH:  0.0  D: 229.6
```
Often the position does not change for several seconds.

DJRaov, 2024-05-19:
> Just before launch the "Sats" field drops to 8, then goes up again. Would make sense, maybe it was moved before release.

Sounds quite likely.
I also caught today's sonde full descent: [https://ub8qbd.satdump.org/files/mrz-n1_descent.wav](https://ub8qbd.satdump.org/files/mrz-n1_descent.wav). If tomorrow morning's launch is as predicted by Sondehub, I will go pick both it and this evening's launches up tomorrow.

rs1729, 2024-05-19:
Now `mrz2mod_test.c` -> `mp3h1mod.c`:
https://github.com/rs1729/RS/tree/master/demod/mod/
I think it is good enough for the next  auto_rx beta.

darksidelemm, 2024-05-20:
This is now in the testing branch.

rs1729, 2024-05-21:
@DJRaov 
The latest MRZ in your area had serial numbers 3xx-3xx, don't know if you could recover one of them. I wonder if they were of the older type, do you have a short recording of 310611-348069 or 31105-348265?
And how about the new type, e.g. 400566-407535 or 401430-407659? Maybe you have a picture, maybe there is a different name for this version on the radiosonde we could use as "subtype". Probably there is also a different GNSS chip inside.

Unfortunately there are some false decodings [401430-407659](https://testing.v2.sondehub.org/#!mt=Mapnik&mz=10&qm=12h&mc=55.60124,64.86534&f=401430-407659&q=401430-407659) with RDZ_TTGO I believe. Sometimes a decoding error with a false SN, or sometimes mixed up with another MRZ. I think that was an older problem, probably fixed in a recent update.

DJRaov, 2024-05-21:
> The latest MRZ in your area had serial numbers 3xx-3xx, don't know if you could recover one of them. 

Nope, sadly, already marked it as not recovered on Sondehub. Another 3xx sonde landed not too far away though, hoping I can pick it up this thursday.
@ub1qbm picked up a 305497 last week, it had the U-Blox GNSS on it:
[image]
I could have him record it for you if you want to.

> And how about the new type, e.g. 400566-407535 or 401430-407659?

I did recover the 400566 sonde, but by sheer, excuse my language, retardation on my side I have erased the FW while trying to dump it.

> Maybe you have a picture, maybe there is a different name for this version on the radiosonde we could use as "subtype". Probably there is also a different GNSS chip inside.

Yep, took pics of the insides: 
[image]
[image]
There is indeed a different GNSS chip, seems like Radiy ran out of U-Bloxes. People seem to be calling it N1-SIM68.

> Unfortunately there are some false decodings [401430-407659](https://testing.v2.sondehub.org/#!mt=Mapnik&mz=10&qm=12h&mc=55.60124,64.86534&f=401430-407659&q=401430-407659) with RDZ_TTGO I believe.

I noted down the coordinates so I can get it some time later (that area had a flooding last month and I got stuck in the mud yesterday trying to recover it), so no worries about it.

rs1729, 2024-05-22:
Thanks!
Is this still called MRZ-N1?

DJRaov, 2024-05-22:
Well, since only the GNSS module changed and nothing else, I don't see why Radiy would rename the model.

ub1qbm, 2024-05-23:
I'd like to add to that. We jokingly named the new probes with a miniature GPS receiver “MRZ-N1-SIM68” because of the visual similarity with GSM communication modules SIM68. For a number of reasons they work “worse” than the previous probes, where the positioning system was based on U-BLOX modules. This is just my opinion, nothing more.
On the left, “MRZ-N1-SIM68.” On the right, “MRZ-N1”
[image]

ub1qbm, 2024-05-23:
I would like to add that “MRZ-N1-SIM68” probes have become labeled with a sticker that reads: “Meteorological radiosonde for measuring air temperature and humidity. Used in Roshydromet network. No danger to humans and the environment. Manufactured in Russia.”
[image]

DJRaov, 2024-06-27:
Been testing the code in the wild for a good while now, works flawlessly. Only thing lacking is voltage readout, but I'll see if I can find it myself. Thanks for your hard work everyone, closing the issue for now.

closed 2024-06-27 by DJRaov

ub1qbm, 2024-09-08:
+rep bro