projecthorus/radiosonde_auto_rx #881: Weathex WxR-301 PN9 variant

darksidelemm, 2024-05-30:
Need to rebase dft_detect and the wxr decoder, and add a new sonde type which uses a 60 kHz tone spacing.

https://github.com/rs1729/RS/issues/56

rs1729, 2024-06-03:
Well, the wide bandwidth is a bit of a problem, the decoder chain is more suitable for modern radiosondes with narrow bandwidths I guess. And not much live data for testing  with auto_rx until now.
I managed to make some recordings, I will update the FM demodulator, this should improve the decoding.

Another issue could be the detection and the center frequency, because there is no frequency correction for this decoder. Taking RS41 as a reference, I observed center frequencies of the two Weathex radiosondes at 403.698 MHz and 404.198 MHz, i.e. 2kHz below the nominal frequency. (Didn't noticed frequency drift between 8-17km.) You have 404.203 MHz, i.e. 5 kHz offset. Without frequency correction that is quite a lot. Was the center frequency at lower altitudes higher?
I believe peak detection in auto_rx is not optimized for this wide signals with two peaks far apart. Maybe it is better to set the center frequency for Weathex manually, if possible.

Joanvoo, 2024-06-03:
Hi, 
The center frequency at lower altitudes are the same.
I try manually more frequenties a little bit higher and lower , but has no no effect.
signal was very strong here.
But the decoder works very good i have more then 9000 [OK] and 100 [NO] frames with a rtl sdr dongle

darksidelemm, 2024-06-03:
If you're using auto_rx, then by default it should be using the fsk_demod chain, which is set up to do centre frequency tracking, and also has the mask-based frequency estimator enabled which appears to have been working well in Malaysia.  I'm using the initial information of 60 kHz tone spacing - is it exactly 60 kHz? 

Can you enable the save_decode_iq option and get an IQ recording from auto_rx? That will help with tweaking the demodulator settings.

I don't currently have a configuration setting which will switch away from fsk_demod and use @rs1729's demodulator chain, but I can add one.

rs1729, 2024-06-03:
Here the filter is 50000 Hz wide, touching both peaks:
[image]

rs1729, 2024-06-03:
So if fsk_demod is used, changes in iq_dec would not help. But even if the rtl_fm/iq_dec decoding chain would be used, I'm afraid that rtl_fm does not filter much when decimating to 96k.  The FM demodulation in iq_dec needs good filtering though.

If fsk_demod was estimating the center frequency as 404.203 MHz, why was it 5kHz off?

rs1729, 2024-06-03:
Here the other WxR-301D, 404.2MHz. (RS41 reference at 403.000MHz no offset)
[image]

darksidelemm, 2024-06-03:
Ahh OK, if the peaks are 50 kHz apart then my setting of 60 kHz in the demodulator will cause issues for sure! 

I've just merged in -beta15, which brings the mask estimator back to 50 kHz. Please let me know if that helps!

rs1729, 2024-06-03:
Yes, fsk_demod with 50k peak-to-peak works good!
No need to use iq_dec.

closed 2024-08-31 by darksidelemm