projecthorus/radiosonde_auto_rx #888: KA9Q Server cmd options pcmcat

papasven, 2024-06-14:
Hi

I'm currently trying testing branch with ka9q way. There is a problem with pcmcat command line options here in ka9q.py.
With "-2" it doesn't work, without is ok.
```
    # pcmcat -2 -s 404090000 sonde-pcm.local
    _cmd = (
        f"pcmcat -2 "
        f"-s {int(frequency)} "
        f"{_pcm_host} |"
    )
```
```
pcmcat -V
KA9Q Multichannel SDR pcmcat.c last modified Sun May 12 11:23:09 2024
Copyright 2023, Phil Karn, KA9Q. May be used under the terms of the GNU Public License
```
something was changed in pcmcat https://github.com/ka9q/ka9q-radio/commit/4d017eb2cfff9fec2733ddc87b9da191600c9479

darksidelemm, 2024-06-14:
I expected this would probably happen, with @ka9q doing lots of of development on ka9q-radio, it was likely that my initial support would break at some point.

I'll talk to Phil and find out what the current correct approach is and try and test.

When support is finally 'released', I'm probably going to have to recommend users check-out and compile a particular commit to avoid this kind of thing happening.

papasven, 2024-06-14:
Ok, would like to remove the spy server someday. ;)
First manual tests with KA9Q worked very well, the pc is no longer sweating.

darksidelemm, 2024-06-14:
Indeed! The performance increase is quite significant, and I would like to move away from spyserver at all of my sites also. However I'm still waiting for spectrum output support from @ka9q before I can complete this... For now it's just fixed frequencies...

darksidelemm, 2024-06-16:
I've fixed the issue with the `-2` option in the testing branch now. I've also made some updates to the [setup notes](https://github.com/projecthorus/radiosonde_auto_rx/wiki/KA9Q%E2%80%90Radio-Notes) since some additional dependencies are required for the latest commit of ka9q-radio.

closed 2024-06-16 by darksidelemm