projecthorus/radiosonde_auto_rx #902: problems with table options!

laarhoven1, 2024-07-25:
i have the table options not in one line any more how to fix this..
why is that it only on my android moblie with google chrome appversion 126.0.6478.186  / Android 14 
[image]

darksidelemm, 2024-07-26:
Unsure sorry! What happens if you turn your phone into landscape mode?

laarhoven1, 2024-07-26:
> Unsure sorry! What happens if you turn your phone into landscape mode?

Same nothing change same issue..
I all reddy did that.

laarhoven1, 2024-07-26:
`
[image]
`

laarhoven1, 2024-07-26:
[image]

closed 2025-03-08 by darksidelemm