projecthorus/radiosonde_auto_rx #916: no port showing in docker PS. but the container is running.

kaunix, 2024-10-08:
when I try to start radiosonde_auto_rx I get the following output:

docker run -d --privileged -v $(pwd)/log:/radiosonde_auto_rx/auto_rx/log/ -p 5000:5000 ghcr.io/projecthorus/radiosonde_auto_rx:latest
6ac569d4335d6b12f4bfbcc80d3545c3639ee9cd7103b10741001aa250657ab4
docker: Error response from daemon: driver failed programming external connectivity on endpoint friendly_antonelli (99acf304ff1e8f419be7324fad5895201fb60092df8e4c3efd2c40e61c9f7546): failed to bind port 0.0.0.0:5000/tcp: Error starting userland proxy: error while calling PortManager.AddPort(): conflict with ID 1.

not sure if this a docker problem or a container problem.

kaunix, 2024-10-08:
I believe that command was lifted directly from your wiki on how to update the container. So if it's wrong? How do YOU start the container ?

kaunix, 2024-10-08:
well I've completely wiped the pi. installe Rasbian bookworm (it needed it anyway) reinstalled docker and radiosonde_auto_rx per the wiki and got the exact same thing. 

:~ $ netstat -tnlp
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -
tcp6       0      0 :::22                   :::*                    LISTEN      -

:~ $ docker ps
CONTAINER ID   IMAGE                                            COMMAND                  CREATED         STATUS                          PORTS     NAMES
168ab4012902   ghcr.io/projecthorus/radiosonde_auto_rx:latest   "/usr/bin/tini -- py…"   8 minutes ago   Restarting (1) 25 seconds ago             radiosonde_auto_rx

Any Ideas?

kaunix, 2024-10-08:
and from root:

sudo netstat -tnlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      705/sshd: /usr/sbin
tcp6       0      0 :::22                   :::*                    LISTEN      705/sshd: /usr/sbin

darksidelemm, 2024-10-08:
I start the container as per the instructions in the wiki, here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Docker
```
docker run \
  -d \
  --name radiosonde_auto_rx \
  --restart="always" \
  --device=/dev/bus/usb \
  --network=host \
  -v ~/radiosonde_auto_rx/station.cfg:/opt/auto_rx/station.cfg:ro \
  -v ~/radiosonde_auto_rx/log/:/opt/auto_rx/log/ \
  ghcr.io/projecthorus/radiosonde_auto_rx:latest
  ```

darksidelemm, 2024-11-01:
Any updates on this?

closed 2024-11-16 by darksidelemm