projecthorus/radiosonde_auto_rx #938: Scan Results in WebInterface with ka9q Server inkomplete?

whallmann, 2024-11-14:
Running autorx 1.8.0 beta 2 with ka9q server.
ka9q server frequency range set to 406.6 as described. Scan grafic shows a stepwise increasing of noise, then drops after 405.5 down.
Some tests now with 407m0 in ka9q shows this grafic. 
[image]
Now 406 MHz get in reach for sonde at 405.7 but this drop down of level lower 401 dont lock good.
Maybe there are some tweeks to solve this? Does someone needs more details?
cu Wolf DF7PN

ke5gdb, 2024-11-14:
Can you post your radiod@.conf file?

ke5gdb, 2024-11-14:
The Airspy Mini is only capable of sampling 6MHz, and with the tuner set at 407MHz, the lower cut-off is 401MHz. Everything is working as designed. You just ran out of bandwidth on that receiver.

whallmann, 2024-11-14:
Thats clear, but not the problem.
First is the scan in autorx. This should cut the beginning down to the left
edge of 401. Its bad to see the baseline because it increases unnessesery
the vertical scala.
Second: if i try to set in the ka9q conf the max frequency to 406m9 level
then the scale jumps a MHz left not only the 0.1 MHz I changed downwards.
So there is something to finetune the cooperation between ka9q server and
autorx. Thats what I mean.

darksidelemm, 2024-11-14:
To clarify - The airspy Mini can only provide **5 MHz** of spectrum, not 6 MHz. 

What was your ka9q-radio settings, and what were your auto_rx min/max settings when producing the picture in the first post? If i remember correctly, the min/max settings will also clip what is shown in the scan plot.

whallmann, 2024-11-15:
> What was your ka9q-radio settings, and what were your auto_rx min/max settings when producing the picture in the first post? If i remember correctly, the min/max settings will also clip what is shown in the scan plot.

The ka9q config is uploaded in post #3  (frequency = 407m0)
autorx_station.cfg (snip)
# Minimum and maximum search frequencies, in MHz.
# ....
# Note - when using either a KA9Q or SpyServer as the SDR backend, the frequency scan limits are set by the
# Server.
in_freq = 402.0
max_freq = 406.0

whallmann, 2024-11-15:
I am very confuses about the settings.  Sometimes values show differences in the scan plot, sometimes not. 
In autorx the in_freq 402. should cut all below it, but if you see in the harccopy above, it wont.

whallmann, 2024-11-15:
Scan plot in 1.8.0 BETA 3 changes to this picture now.   I dont think this was expected?!
[image]

whallmann, 2024-11-15:
Simple another try to see how confusing this scan plot is.
ka9q: frequency  406m5
autorx: low 402.0   high 405.8
[image]
(1) low area is now filled with something
(2) high area looks funny this time.

darksidelemm, 2024-11-15:
That you have set auto_rx to a low of 402 MHz and a high or 405.8 MHz and you are getting data that extends down to 400.5 MHz makes me wonder if something else is not set correctly. 

Please try the following settings *exactly*, as I want to at least confirm you can replicate my setup:
radiod@airspymini-autorx.conf
```
[global]
hardware = airspy
mode = fm
status = sonde.local
iface = lo
ttl = 0  
data = sonde-pcm.local

[airspy]
device = airspy
description = "auto_rx"
# The airspy frequency must be set at least 600 kHz above the highest frequency we wish to observe.
# As the Airspy Mini can only observe ~5 MHz of spectrum at once, we can set this to either
# 405.6 MHz (405m6, to cover 400-405 MHz) or 406.6 MHz (406m6, to cover 401-405 MHz) depending on where sondes are prevalent in your local area.
frequency = 405m6

[telemetry]
freq = "401m50"

[manual-400]
freq = 0
ttl = 0
```

settings in station.cfg:
```
min_freq = 400.05
max_freq = 404.0
```

At my site, this gives me a plot that looks like this:
[image]

whallmann, 2024-11-15:
Hi Mark,
as you requested, with the settings of this values from last msg following scan plot is displayed:
[image]

whallmann, 2024-11-15:
Aua, deteced a typo in the station.cfg  - While editing so frequently, the name of the "min_freq" was "in_freq".  Now with the "min_freq" value name the lower limit is correct.
Just checked:
ka9q : 406m6
autorx:   min_value = 402.05 / max_value = 406.0
looks like this:
[image]

darksidelemm, 2024-11-15:
Well, I'm at at a loss as to why your spectrum looks so weird. I still think some setting is incorrect... 

Next things to look at. Turn on debug logging, and enable system log so we can see exactly what commands are being sent. These are in this section of the station.cfg file: https://github.com/projecthorus/radiosonde_auto_rx/blob/testing/auto_rx/station.cfg.example#L454

In your log directory should be a file with a name like 'log_power_KA9Q-01.csv' - can you please copy this out and upload it either here, or somewhere else. This is the raw output from ka9q-server.

argilo, 2024-11-16:
The bad behaviour turned out to be due to a bug in ka9q-radio. I proposed a fix here: https://github.com/ka9q/ka9q-radio/pull/65#issuecomment-2480243690

Auto-RX could work around the problem like so:
```diff
diff --git a/auto_rx/autorx/sdr_wrappers.py b/auto_rx/autorx/sdr_wrappers.py
index bece143..fa176f1 100644
--- a/auto_rx/autorx/sdr_wrappers.py
+++ b/auto_rx/autorx/sdr_wrappers.py
@@ -779,6 +779,7 @@ def get_power_spectrum(
         _ssrc = f"{round(_center_freq / 1000)}03"
 
         _powers_cmd = (
+            f"LANG=C "
             f"{_timeout_cmd} {ka9q_powers_path} "
             f"{sdr_hostname} "
             f"-f {_center_freq} "
```

argilo, 2024-11-16:
This should now be fixed on the Auto-RX testing branch.

whallmann, 2024-11-16:
OK, done this with the debug flags. Results in two files and one pic with scan plot
At this time the frequencys are:  ka9q = 407m1,  autorx: low = 401.9 upper 406.0
Last pull as argilo mentioned was installed before.  
[image]
[20241116-095816_system.log](https://github.com/user-attachments/files/17784237/20241116-095816_system.log)
[log_power_KA9Q-01.csv](https://github.com/user-attachments/files/17784238/log_power_KA9Q-01.csv)

darksidelemm, 2024-11-16:
At this stage I expect this is probably just whatever is present, and might be something to do with the airspy itself... 

I'd suggest trying out the AGC mode  (e.g. don't specify a gainstep) and see if that still works well enough even with your separate preamp. 

I think we can probably close this issue out now.

PR #941 (2024-11-16): Work around ka9q-radio powers locale bug
Fixes #938.

Forcing the C locale when running `powers` will prevent its CSV output from being corrupted.

This workaround can be removed once https://github.com/ka9q/ka9q-radio/pull/65 has been merged and enough time has passed that nobody is running buggy versions.
closed 2024-12-14 by darksidelemm