projecthorus/radiosonde_auto_rx #944: Trouble when not running from the venv

gariac, 2024-11-20:
Linux framework13 6.11.7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.11.7-1 (2024-11-09) x86_64 GNU/Linux
Version: 2024-11-19 23:08:46,848 INFO:Version - Local Version: 1.7.4 - Up to date!
I downloaded from githum on 11/19/2024

I installed the program is /usr/local/src rather than my home directory.

Here is the invocation from venv:
`(venv) root@framework13:/usr/local/src/radiosonde_auto_rx/auto_rx# /usr/local/src/radiosonde_auto_rx/auto_rx/venv/bin/python3 /usr/local/src/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
2024-11-19 23:21:34,253 INFO:Reading configuration file...
2024-11-19 23:21:34,254 WARNING:Config - Web Password not set, disabling web control
2024-11-19 23:21:38,099 INFO:Config - Tested RTLSDR #0 OK
2024-11-19 23:21:38,100 INFO:Started Flask server on http://0.0.0.0:5000
2024-11-19 23:21:38,101 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2024-11-19 23:21:38,101 INFO:OziMux - Started OziMux / Payload Summary Exporter
2024-11-19 23:21:38,102 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2024-11-19 23:21:38,632 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2024-11-19 23:21:38,668 INFO:Version - Local Version: 1.7.4 - Up to date!
2024-11-19 23:21:38,668 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2024-11-19 23:21:42,601 INFO:Scanner (RTLSDR 0) - Starting Scanner Thread
2024-11-19 23:21:42,602 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2024-11-19 23:22:04,112 INFO:Scanner (RTLSDR 0) - Detected peaks on 1 frequencies (MHz): [403.2]
2024-11-19 23:22:21,714 INFO:Scanner (RTLSDR 0) - Running frequency scan.
2024-11-19 23:22:43,234 INFO:Scanner (RTLSDR 0) - Detected peaks on 1 frequencies (MHz): [403.2]
^C2024-11-19 23:22:46,835 INFO:Web - Flask Server Shutdown.
2024-11-19 23:22:46,835 INFO:Starting shutdown of all threads.
2024-11-19 23:22:46,835 INFO:Scanner (RTLSDR 0) - Waiting for current scan to finish...
2024-11-19 23:22:46,975 INFO:Scanner (RTLSDR 0) - Scanner Thread Closed.
2024-11-19 23:22:47,177 INFO:Telemetry Logger - Stopped Telemetry Logger Thread.
2024-11-19 23:22:48,677 INFO:Sondehub Uploader - Stopped Sondehub Uploader Thread.

`
As you can see the program runs from within the venv. However for me to make this a service, I need to invoke the software from systemd. To test this, it is sufficient just to use the same command line from a terminal, that is not in venv:

`
root@framework13:/etc/systemd/system# /usr/local/src/radiosonde_auto_rx/auto_rx/venv/bin/python3 /usr/local/src/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
Invalid logging path, using default. Does the folder exist?
2024-11-19 23:27:03,452 CRITICAL:Binary dft_detect does not exist - did you run build.sh?
root@framework13:/etc/systemd/system# 
`
I took a look at 
[https://docs.python.org/3/library/venv.html#how-venvs-work](url)
and your instructions should work. I don't know if you need the shebangs.

Note I realize the supplied systemd server won't work. However I need to get the simple command line invocation working first.

TheSkorm, 2024-11-20:
I suspect autorx expects you to execute it from the directory its installed, you'll need to change the directory into the /usr/local/src/radiosonde_auto_rx/ folder for it to work. This isn't really a supported way of running autorx.

gariac, 2024-11-20:
Well ultimately the software needs to be invoked from /systemd/system to create a service, so I suspect I don't understand your reply. However:
`
root@framework13:/usr/local/src/radiosonde_auto_rx# /usr/local/src/radiosonde_auto_rx/auto_rx/venv/bin/python3 /usr/local/src/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
Invalid logging path, using default. Does the folder exist?
2024-11-19 23:39:58,833 CRITICAL:Binary dft_detect does not exist - did you run build.sh?
root@framework13:/usr/local/src/radiosonde_auto_rx# 
`

TheSkorm, 2024-11-20:
I was one folder off sorry - that needs to be `/usr/local/src/radiosonde_auto_rx/autorx`

You can see here that the systemd unit file is configured with a working directory 

https://github.com/projecthorus/radiosonde_auto_rx/wiki#45-automatic--continuous-operation-via-systemd

gariac, 2024-11-20:
Thanks. That gives me something to ponder. (If you are in the US, something to chew on.)

`
root@framework13:/usr/local/src/radiosonde_auto_rx/auto_rx# pwd
/usr/local/src/radiosonde_auto_rx/auto_rx
root@framework13:/usr/local/src/radiosonde_auto_rx/auto_rx# /usr/local/src/radiosonde_auto_rx/auto_rx/venv/bin/python3 /usr/local/src/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
2024-11-19 23:45:56,340 INFO:Reading configuration file...
2024-11-19 23:45:56,340 WARNING:Config - Web Password not set, disabling web control
2024-11-19 23:46:00,216 INFO:Config - Tested RTLSDR #0 OK
2024-11-19 23:46:00,217 INFO:Started Flask server on http://0.0.0.0:5000
2024-11-19 23:46:00,217 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2024-11-19 23:46:00,218 INFO:OziMux - Started OziMux / Payload Summary Exporter
2024-11-19 23:46:00,218 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2024-11-19 23:46:00,726 INFO:Version - Local Version: 1.7.4 - Up to date!
2024-11-19 23:46:00,726 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2024-11-19 23:46:00,764 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
^C2024-11-19 23:46:02,150 INFO:Web - Flask Server Shutdown.
2024-11-19 23:46:02,150 INFO:Starting shutdown of all threads.
2024-11-19 23:46:02,150 ERROR:Error stopping task - 'NoneType' object has no attribute 'stop'
2024-11-19 23:46:02,219 INFO:Telemetry Logger - Stopped Telemetry Logger Thread.
2024-11-19 23:46:05,766 INFO:Sondehub Uploader - Stopped Sondehub Uploader Thread.
`

gariac, 2024-11-20:
I did something wrong with the quotes so my reply looks like a mess. Anyway changing the directory as indicated solved the problem. I probably need to use  symbolic link in that directory,

closed 2024-11-20 by TheSkorm