projecthorus/radiosonde_auto_rx #945: AutoRX V1.0.8beta3 crashed with KA9Q Server

Mooni123, 2024-11-21:
Hello,

I started testing the new KA9q-server usage with the Auto-RX V1.0.8beta3.
As hardware, I use the AirSpyMini.

I have the following infos in logfile:

`
2024-11-21 20:18:17,816 ERROR:Decoder (KA9Q sonde.local) RS41 402.700 - RX Timed out.
2024-11-21 20:18:18,817 INFO:Decoder (KA9Q sonde.local) RS41 402.700 - Closed decoder subprocess.
2024-11-21 20:18:19,792 DEBUG:KA9Q - Closing Channel for  @ None Hz.
Traceback (most recent call last):
  File "/home/pi/radiosonde_auto_rx_ka9q/auto_rx/auto_rx.py", line 1119, in <module>
    main()
  File "/home/pi/radiosonde_auto_rx_ka9q/auto_rx/auto_rx.py", line 1095, in main
    clean_task_list()
  File "/home/pi/radiosonde_auto_rx_ka9q/auto_rx/auto_rx.py", line 448, in clean_task_list
    shutdown_sdr(config["sdr_type"], _task_sdr)
  File "/home/pi/radiosonde_auto_rx_ka9q/auto_rx/autorx/sdr_wrappers.py", line 261, in shutdown_sdr
    ka9q_close_channel(sdr_hostname, frequency, scan)
  File "/home/pi/radiosonde_auto_rx_ka9q/auto_rx/autorx/ka9q.py", line 85, in ka9q_close_channel
    f"--ssrc {round(frequency / 1000)}{ssrc} "
                    ~~~~~~~~~~^~~~~~
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
Main Loop Error - unsupported operand type(s) for /: 'NoneType' and 'int'
`

It seems it crash AutoRx on closing a KA9Q-Server channel without a frequency / ssrc.

Has anyone else such a problem?

Thanks for any help,

Mooni123

darksidelemm, 2024-11-21:
It looks like you've got debug logging enabled - can you get us more of the log, so we can see how that decoder was started up? It's certainly weird that it should get into this state.

argilo, 2024-11-24:
@Mooni123 Have you modified Auto-RX? The line numbers in your stack trace don't seem to match up with the official version.

Mooni123, 2024-11-27:
Hello together,

I checked out actual testing branch again and the problem is gone now.
So, I think there was a problems on my side.
Yes, I made some modifications on Maps display (added selection of additional map layers).

So, we can close this issue here.

Sorry.
Greetings,
Mooni123

closed 2024-11-27 by Mooni123