projecthorus/radiosonde_auto_rx #946: Auto-block Peaks that do not have a valid sonde decode after X decodes

darksidelemm, 2024-11-24:
An idea from Zigi's fork of auto_rx (which has other geographic-specific features which will not be added into auto_rx for various reasons...).

If a peak shows up in the scan result more than X times in a row, with no valid sonde detection, add it to the temporary block list (which locks out that peak for 2 hours by default). This should help with reducing scan time on 'birdies' or other non-radiosonde signals that are always present. 

The value for X will need to be tuned to avoid sondes that might just be starting to become decodable to be locked out.

darksidelemm, 2024-12-07:
The other issue we see is non-radiosonde signals (in my case, trunked radio network channel) being detected as iMet sondes, and auto_rx starting up a decoder. The above approach wouldn't lock these out.

I guess we could also add something in that temporarily locks out a frequency if it's been detected as an iMet, but has timed out multiple times?

helioshk, 2024-12-17:
Or make dft_detect smarter, so it can recognize unmodulated tones or trunked radio?