projecthorus/radiosonde_auto_rx #948: Evaluate using multi-of-200Hz sample rates for M10

darksidelemm, 2024-11-27:
The issue:

- M10's have a weird baud rate of 9616 Hz. 
- fsk_demod wants a sample rate that is an integer multiple of the baud rate. Currently we use 5x, so 48080 Hz.
- ka9q-radio only supports sample rates that are a multiple of 200 Hz, so it rounds 48080 Hz to 48000 Hz.

Options to deal with this:
- Make use of the tsrc fractional resampler ( https://github.com/projecthorus/radiosonde_auto_rx/blob/master/utils/tsrc.c ), include this in the build process, and use this in the function that gets an KA9Q IQ command if the sample rate isn't a multiple of 200.
- Test the performance impact of running fsk_demod at a sample rate and baud rate that does not precisely match the M10, e.g
  - 9600 baud, 48000 Hz
  - 9620 baud, 96200 Hz
  - This can be tested using the existing IQ samples and the scripts in https://github.com/projecthorus/radiosonde_auto_rx/tree/master/auto_rx/test

darksidelemm, 2024-11-27:
Some quick data.
(filename, packets received, time to process)

Looks like not much performance difference between the 3 approaches!

```
(venv) darkside@Marks-MacBook-Pro test % python test_demod.py -f "generated/m10_96k_float_*" -m m10_fsk_demod_soft_centre
Command: cat generated/m10_96k_float_05.0dB.bin | csdr convert_f_s16 | ../tsrc - - 0.50083333333 -c | ../fsk_demod --cs16 -b -10000 -p 5 -u 10000 -s --stats=5 2 48080 9616 - - 2>stats.txt |../m10mod --json --softin -i -vvv 2>/dev/null| grep aprsid | wc -l
m10_96k_float_05.0dB.bin, 0, 3.612
m10_96k_float_05.5dB.bin, 0, 0.580
m10_96k_float_06.0dB.bin, 0, 0.637
m10_96k_float_06.5dB.bin, 0, 0.666
m10_96k_float_07.0dB.bin, 1, 0.621
m10_96k_float_07.5dB.bin, 1, 0.586
m10_96k_float_08.0dB.bin, 11, 0.619
m10_96k_float_08.5dB.bin, 40, 0.623
m10_96k_float_09.0dB.bin, 69, 0.621
m10_96k_float_09.5dB.bin, 88, 0.640
m10_96k_float_10.0dB.bin, 99, 0.632
m10_96k_float_10.5dB.bin, 113, 0.672
m10_96k_float_11.0dB.bin, 117, 0.599
m10_96k_float_11.5dB.bin, 120, 0.591
m10_96k_float_12.0dB.bin, 120, 0.744
m10_96k_float_12.5dB.bin, 118, 0.607
m10_96k_float_13.0dB.bin, 120, 0.583
m10_96k_float_13.5dB.bin, 119, 0.612
m10_96k_float_14.0dB.bin, 119, 0.817
m10_96k_float_14.5dB.bin, 120, 0.603
m10_96k_float_15.0dB.bin, 119, 0.668
m10_96k_float_15.5dB.bin, 120, 0.736
m10_96k_float_16.0dB.bin, 120, 0.656
m10_96k_float_16.5dB.bin, 120, 0.749
m10_96k_float_17.0dB.bin, 120, 0.596
m10_96k_float_17.5dB.bin, 120, 0.619
m10_96k_float_18.0dB.bin, 120, 0.811
m10_96k_float_18.5dB.bin, 120, 0.676
m10_96k_float_19.0dB.bin, 120, 0.658
m10_96k_float_19.5dB.bin, 120, 0.660
m10_96k_float_20.0dB.bin, 120, 0.690
(venv) darkside@Marks-MacBook-Pro test % python test_demod.py -f "generated/m10_96k_float_*" -m m10_fsk_demod_soft_centre_48000
Command: cat generated/m10_96k_float_05.0dB.bin | csdr convert_f_s16 | ../tsrc - - 0.500 -c | ../fsk_demod --cs16 -b -10000 -p 5 -u 10000 -s --stats=5 2 48000 9600 - - 2>stats.txt |../m10mod --json --softin -i -vvv 2>/dev/null| grep aprsid | wc -l
m10_96k_float_05.0dB.bin, 0, 0.616
m10_96k_float_05.5dB.bin, 0, 0.636
m10_96k_float_06.0dB.bin, 0, 0.582
m10_96k_float_06.5dB.bin, 0, 0.580
m10_96k_float_07.0dB.bin, 0, 0.575
m10_96k_float_07.5dB.bin, 0, 0.571
m10_96k_float_08.0dB.bin, 14, 0.570
m10_96k_float_08.5dB.bin, 46, 0.572
m10_96k_float_09.0dB.bin, 71, 0.731
m10_96k_float_09.5dB.bin, 87, 0.925
m10_96k_float_10.0dB.bin, 103, 0.835
m10_96k_float_10.5dB.bin, 113, 0.785
m10_96k_float_11.0dB.bin, 118, 0.771
m10_96k_float_11.5dB.bin, 120, 0.836
m10_96k_float_12.0dB.bin, 120, 0.863
m10_96k_float_12.5dB.bin, 118, 0.871
m10_96k_float_13.0dB.bin, 120, 0.889
m10_96k_float_13.5dB.bin, 120, 0.619
m10_96k_float_14.0dB.bin, 119, 0.604
m10_96k_float_14.5dB.bin, 120, 0.552
m10_96k_float_15.0dB.bin, 119, 0.553
m10_96k_float_15.5dB.bin, 120, 0.569
m10_96k_float_16.0dB.bin, 120, 0.570
m10_96k_float_16.5dB.bin, 120, 0.561
m10_96k_float_17.0dB.bin, 120, 0.555
m10_96k_float_17.5dB.bin, 120, 0.588
m10_96k_float_18.0dB.bin, 120, 0.567
m10_96k_float_18.5dB.bin, 120, 0.592
m10_96k_float_19.0dB.bin, 120, 0.566
m10_96k_float_19.5dB.bin, 120, 0.575
m10_96k_float_20.0dB.bin, 120, 0.559
(venv) darkside@Marks-MacBook-Pro test % python test_demod.py -f "generated/m10_96k_float_*" -m m10_fsk_demod_soft_centre_96200
Command: cat generated/m10_96k_float_05.0dB.bin | csdr convert_f_s16 | ../tsrc - - 1.0020833333333334 -c | ../fsk_demod --cs16 -b -10000 -p 5 -u 10000 -s --stats=5 2 96200 9620 - - 2>stats.txt |../m10mod --json --softin -i -vvv 2>/dev/null| grep aprsid | wc -l
m10_96k_float_05.0dB.bin, 0, 0.711
m10_96k_float_05.5dB.bin, 0, 0.740
m10_96k_float_06.0dB.bin, 0, 0.726
m10_96k_float_06.5dB.bin, 0, 0.705
m10_96k_float_07.0dB.bin, 0, 0.713
m10_96k_float_07.5dB.bin, 3, 0.765
m10_96k_float_08.0dB.bin, 8, 0.721
m10_96k_float_08.5dB.bin, 36, 0.766
m10_96k_float_09.0dB.bin, 67, 0.731
m10_96k_float_09.5dB.bin, 84, 0.711
m10_96k_float_10.0dB.bin, 102, 0.728
m10_96k_float_10.5dB.bin, 112, 0.749
m10_96k_float_11.0dB.bin, 118, 0.757
m10_96k_float_11.5dB.bin, 119, 0.729
m10_96k_float_12.0dB.bin, 120, 0.699
m10_96k_float_12.5dB.bin, 120, 0.722
m10_96k_float_13.0dB.bin, 120, 0.742
m10_96k_float_13.5dB.bin, 120, 0.713
m10_96k_float_14.0dB.bin, 119, 0.733
m10_96k_float_14.5dB.bin, 120, 0.731
m10_96k_float_15.0dB.bin, 119, 0.764
m10_96k_float_15.5dB.bin, 120, 0.702
m10_96k_float_16.0dB.bin, 120, 0.698
m10_96k_float_16.5dB.bin, 120, 0.691
m10_96k_float_17.0dB.bin, 120, 0.696
m10_96k_float_17.5dB.bin, 120, 0.716
m10_96k_float_18.0dB.bin, 120, 0.723
m10_96k_float_18.5dB.bin, 120, 0.707
m10_96k_float_19.0dB.bin, 120, 0.696
m10_96k_float_19.5dB.bin, 120, 0.709
m10_96k_float_20.0dB.bin, 120, 0.714
```

argilo, 2024-11-27:
A couple other possibilities:

* 9616.666 baud, 115400 Hz
* 9616.000 baud, 240400 Hz

darksidelemm, 2024-12-06:
Another test, which emulates the current behaviour, where we tell fsk_demod it's getting 48080 Hz sample rate IQ, but it's actually getting 48000 Hz.

The results are all pretty close to the above tests... Makes me think we leave this one along and worry about tweaking performance if we see reports of decoding issues from KA9Q stations decoding M10s.

```
(venv) darkside@Marks-MacBook-Pro test % python test_demod.py -f "generated/m10_96k_float_*" -m m10_fsk_demod_soft_centre_48000_2
Command: cat generated/m10_96k_float_05.0dB.bin | csdr convert_f_s16 | ../tsrc - - 0.500 -c | ../fsk_demod --cs16 -b -10000 -p 5 -u 10000 -s --stats=5 2 48080 9616 - - 2>stats.txt |../m10mod --json --softin -i -vvv 2>/dev/null| grep aprsid | wc -l
m10_96k_float_05.0dB.bin, 0, 1.341
m10_96k_float_05.5dB.bin, 0, 1.575
m10_96k_float_06.0dB.bin, 0, 1.066
m10_96k_float_06.5dB.bin, 0, 1.029
m10_96k_float_07.0dB.bin, 0, 1.142
m10_96k_float_07.5dB.bin, 0, 1.142
m10_96k_float_08.0dB.bin, 14, 1.047
m10_96k_float_08.5dB.bin, 46, 1.134
m10_96k_float_09.0dB.bin, 71, 1.200
m10_96k_float_09.5dB.bin, 88, 0.812
m10_96k_float_10.0dB.bin, 103, 1.082
m10_96k_float_10.5dB.bin, 113, 1.129
m10_96k_float_11.0dB.bin, 118, 4.283
m10_96k_float_11.5dB.bin, 120, 1.807
m10_96k_float_12.0dB.bin, 120, 1.193
m10_96k_float_12.5dB.bin, 118, 1.064
m10_96k_float_13.0dB.bin, 120, 1.099
m10_96k_float_13.5dB.bin, 120, 1.017
m10_96k_float_14.0dB.bin, 119, 1.248
m10_96k_float_14.5dB.bin, 120, 1.285
m10_96k_float_15.0dB.bin, 119, 0.964
m10_96k_float_15.5dB.bin, 120, 1.453
m10_96k_float_16.0dB.bin, 120, 1.266
m10_96k_float_16.5dB.bin, 120, 1.395
m10_96k_float_17.0dB.bin, 120, 1.619
m10_96k_float_17.5dB.bin, 120, 1.742
m10_96k_float_18.0dB.bin, 120, 1.192
m10_96k_float_18.5dB.bin, 120, 1.154
m10_96k_float_19.0dB.bin, 120, 1.224
m10_96k_float_19.5dB.bin, 120, 1.104
m10_96k_float_20.0dB.bin, 120, 1.136
```

closed 2025-03-08 by darksidelemm