projecthorus/radiosonde_auto_rx #960: Issue with scan plot on 1.8.0

lincdeckard, 2024-12-16:
Hello, after update the scan plot now shows up like this: 

[image]

Any thoughts?

Best,
Carlos

darksidelemm, 2024-12-16:
This is pretty much what is expected. We've removed the section header above the scan plot, and the frequency scale is now cleaned up.

lincdeckard, 2024-12-16:
I mean, the plot isn't correct. It's only noise and seems to be mirrored 3 times. It doesn't correspond to the actual RF spectrum.

darksidelemm, 2024-12-16:
Yes, quite possibly. RTLSDR scanning hasn't changed in this release, so I'm not sure why that would have happened. Perhaps check your RF setup? Maybe try switching back to one of the SDRs and see if that makes a difference?

lincdeckard, 2024-12-16:
Ah that was it, I needed to reboot the raspberry to reset both RTL dongles. It's working fine now, thanks!

closed 2024-12-16 by lincdeckard