projecthorus/radiosonde_auto_rx #965: scrolling

EugenNino, 2024-12-26:
Why doesn't scrolling work in settings mode, I press the down button but it doesn't scroll to see other settings. attached a screenshot
[image]

darksidelemm, 2024-12-26:
I'm not sure the down button ever did that? I think it was there just to indicate that there's more settings below?
You need to use your mouse wheel to scroll down the settings area.

EugenNino, 2025-01-17:
I wanted to say that the laptop's touchpad does not allow this, that is, you can only scroll down either with the mouse wheel or swipe across the screen on your smartphone.

closed 2025-03-08 by darksidelemm