projecthorus/radiosonde_auto_rx #966: SpyServer not receiving anything

Feuerwerko, 2024-12-26:
Hello,
First off, I dont know if this is the right place for this question as its more spyserver related, but I dont know where else I would ask it.
I just set up a spyserver + autorx configuration on my rpi 4 that used to run an RTLSDR.
For the first few hours it seemed to work fine (receving sondes) but earlier today spyserver was no longer able to recognize the airspy mini. Seemed to be a permission issue, after adding myself to the proper group it worked again. No clue why it worked before.
The issue im having now is that the spyserver seems to keep outputting the same iq data. Even if I connect to it via SDR++ and change the frequency it outputs the same FFT. Ive upgraded to the latest version of autorx and the issue seems to persist.
[image]
The tasking and everything is correct, just not the FFT. At the time of this image there was a sonde on 402.7 mhz that I should be able to receive.
This is what SDR++ looks like:
[image]

This is my spyserver config:
```
# SPY Server Configuration File

# TCP Listener
#
bind_host = 0.0.0.0
bind_port = 5555-6666

# List Server in Airspy Directory
#
list_in_directory = 0

# Owner Name
# For example: John Doe L8ZEE
owner_name =

# Owner email
# For example: john@doe.com
owner_email =

# Antenna
# For example: Random Wire/Magnetic Loop/Mini-Whip/Inverted V/etc.
antenna_type =

# Antenna Location
# For example: 48.858332, 2.294560
antenna_location =

# General Description
#
general_description =

# User sessions
#
maximum_clients = 5

# Maximum session duration
# In minutes. 0 for no limit
#
#maximum_session_duration = 30

# Allow clients to retune and change of gain of the device
#
allow_control = 1

# Device Type
# Possible Values:
#   AirspyOne (R0, R2, Mini)
#   AirspyHF+
#   RTL-SDR
#   Auto      (Scans for the first available device)
#
device_type = AirspyOne

# Device Serial Number as 64bit Hex
# For example: 0xDD52D95C904534AD
# A value of 0 will acquire the first available device.
#
device_serial = 0

# Device Sample Rate
# Possible Values:
#   Airspy R0, R2 : 10000000 or 2500000
#   Airspy Mini   :  6000000 or 3000000
#   Airspy HF+    :   768000
#   RTL-SDR       :   500000 to 3200000
# Comment to use the device's default
#
#device_sample_rate = 2500000

# Force 8bit Compression Mode
# The 8bit Compression mode has proven sufficiently good for most streaming use cases.
# Use it to same some internet bandwidth.
#
#force_8bit = 1

# Maximum Bandwidth
# Limits the maximum IQ bandwidth the clients can set
# Recommended value for WFM is 200000
# Recommended value for narrow band modes is 15000
#
#maximum_bandwidth = 15000

# FFT Frames Per Second
#
fft_fps = 20

# FFT Bins
# Bins = 2^fft_bin_bits
#
fft_bin_bits = 16

# Initial Center Frequency
#
initial_frequency = 403000000

# Minimum Tunable Frequency
# Comment if using the device default
#
minimum_frequency = 400600000

# Maximum Tunable Frequency
# Comment if using the device default
#
maximum_frequency = 405400000

# Frequency Correction in PPB
#
#frequency_correction_ppb = 0

# Initial Gain
#
#initial_gain = 15

# RTL-SDR Sampling mode
# Quadrature = 0,
# Direct Sampling I Branch = 1
# Direct Sampling Q Branch = 2
#rtl_sampling_mode = 0

# Converter Offset
# Set to -120000000 to enable the SpyVerter offset
#converter_offset = -120000000

# Bias-Tee
# For AirspyOne only - Useful for LNA's and SpyVerter
#enable_bias_tee = 0

# Input IQ Buffering
#
input_buffer_size_ms = 10
input_buffer_count = 4

# Output Network Buffer Size (in milliseconds)
#
output_buffer_size_ms = 50
```

I dont know if the max, initial and min frequency settings are necessary. They match the frequencys in the autorx configuration. When I first set it up, autorx would always give me errors that 402.7 was out of bounds. This was resolved by adding the min, max and initial frequency settings.

Im quite confused where the spikes in the FFT are coming from, because all my other SDRs dont receive them so its not RFI.

I would appreciate any help!

Feuerwerko, 2024-12-26:
This seems to be the issue. SDR++ receives the signals just fine when I remove the frequency presets:
[image]

Feuerwerko, 2024-12-26:
Okay I just tried the spyserver config from the wiki. Didnt see it before, my mistake.
When I try that config, only one of the tasks crashes and it continues receiving fine.
[image]

ke5gdb, 2024-12-26:
Are you using v1.8.0? If so, try updating to the latest testing version. There was an issue identified a few hours ago pertaining to return code 124 (timeout error) that was patched. 

You should also consider using KA9Q-Radio.

Feuerwerko, 2024-12-26:
Yes, I just updated.
I was thinking of using KA9Q Radio but it seems a little experimental and poorly documented.
Are there any big advantages that KA9Q radio has over spyserver?

ke5gdb, 2024-12-26:
Check the install notes here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/KA9Q%E2%80%90Radio-Setup-Notes

It's just a better platform. It's more efficient, actively developed, and open source. However, before you consider making the change, I'm curious if updating to the latest testing version fixed the issue.

Feuerwerko, 2024-12-26:
Switching to testing did indeed work. No errors now.
Im gonna try switching to KA9Q and see if it works.

closed 2024-12-26 by Feuerwerko