projecthorus/radiosonde_auto_rx #967: KA9Q Radio couldnt contact sonde.local

Feuerwerko, 2024-12-26:
Hello,
Ive just finished setting up ka9q radio as per this wiki page: https://github.com/projecthorus/radiosonde_auto_rx/wiki/KA9Q%E2%80%90Radio-Setup-Notes

Im having an issue where sonde.local seemingly doesnt exist.
When I run `control sonde.local` as suggested in the wiki, I get this output: `resolve_mcast getaddrinfo(host=sonde.local, port=(null)): Name or service not known. Retrying.`

AutoRX also doesnt seem to be able to contact sonde.local. Output:
```
Dec 26 19:36:24 DietPi sudo[7071]:     root : PWD=/home/dietpi/radiosonde_auto_rx/auto_rx ; USER=root ; COMMAND=/home/dietpi/radiosonde_auto_rx/auto_rx/venv/bin/python3 /home/dietpi/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
Dec 26 19:36:24 DietPi sudo[7071]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Dec 26 19:36:26 DietPi auto_rx[7072]: 2024-12-26 19:36:26,081 INFO:Reading configuration file...
Dec 26 19:36:26 DietPi auto_rx[7072]: 2024-12-26 19:36:26,088 WARNING:Config - Web Password not set, disabling web control
Dec 26 19:37:26 DietPi auto_rx[7072]: 2024-12-26 19:37:26,096 CRITICAL:KA9Q (sonde.local) - tune call failed with a timeout. Is the server running?
Dec 26 19:37:26 DietPi auto_rx[7072]: 2024-12-26 19:37:26,096 CRITICAL:Config - Could not contact KA9Q Server sonde.local:5555. Exiting. <- i assume the sonde.local:5555 is an error in the logging code
Dec 26 19:37:26 DietPi auto_rx[7072]: 2024-12-26 19:37:26,097 CRITICAL:Error in configuration file! Exiting...
Dec 26 19:37:26 DietPi sudo[7071]: pam_unix(sudo:session): session closed for user root
Dec 26 19:37:26 DietPi systemd[1]: auto_rx.service: Main process exited, code=exited, status=1/FAILURE
Dec 26 19:37:26 DietPi systemd[1]: auto_rx.service: Failed with result 'exit-code'.
Dec 26 19:37:26 DietPi systemd[1]: auto_rx.service: Consumed 2.098s CPU time.
Dec 26 19:39:26 DietPi systemd[1]: auto_rx.service: Scheduled restart job, restart counter is at 8.
Dec 26 19:39:26 DietPi systemd[1]: Stopped auto_rx.service - auto_rx.
Dec 26 19:39:26 DietPi systemd[1]: auto_rx.service: Consumed 2.098s CPU time.
Dec 26 19:39:26 DietPi systemd[1]: Started auto_rx.service - auto_rx.
```

My KA9Q Radio config is exactly the config from the airspy mini section in the wiki. The radio daemon itself seems to boot up normally, it just cant be reached.
```
$ sudo journalctl -f -n 100 -u radiod@airspymini-autorx
Dec 26 19:06:12 DietPi radiod[6270]: Established under name 'airspymini-autorx'
Dec 26 19:06:12 DietPi radiod[6268]: Established under name 'airspymini-autorx'
Dec 26 19:06:12 DietPi radiod[6271]: Established under name 'sonde-pcm.local'
Dec 26 19:06:12 DietPi radiod[6269]: Established under name 'sonde-pcm.local'
Dec 26 19:06:12 DietPi radiod[6279]: Established under name 'telemetry'
Dec 26 19:06:12 DietPi radiod[6278]: Established under name 'sonde.local'
Dec 26 19:06:12 DietPi radiod[6280]: Established under name 'sonde-pcm.local'
Dec 26 19:06:12 DietPi radiod[6282]: Established under name 'manual-400'
Dec 26 19:06:12 DietPi radiod[6277]: Established under name 'auto_rx'
Dec 26 19:06:12 DietPi radiod[6283]: Established under name 'sonde-pcm.local'
Dec 26 19:09:04 DietPi radiod[6266]: Received signal 15, exiting
Dec 26 19:09:04 DietPi radiod[6269]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6277]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi systemd[1]: Stopping radiod@airspymini-autorx.service - airspymini-autorx radio receiver...
Dec 26 19:09:04 DietPi radiod[6278]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6270]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6283]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6282]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6271]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6280]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6279]: Got SIGTERM, quitting.
Dec 26 19:09:04 DietPi radiod[6268]: Got SIGTERM, quitting.
Dec 26 19:09:05 DietPi systemd[1]: radiod@airspymini-autorx.service: Deactivated successfully.
Dec 26 19:09:05 DietPi systemd[1]: Stopped radiod@airspymini-autorx.service - airspymini-autorx radio receiver.
Dec 26 19:09:05 DietPi systemd[1]: radiod@airspymini-autorx.service: Consumed 1min 43.564s CPU time.
Dec 26 19:09:05 DietPi systemd[1]: Started radiod@airspymini-autorx.service - airspymini-autorx radio receiver.
Dec 26 19:09:05 DietPi radiod[6475]: KA9Q Multichannel SDR main.c last modified Thu Dec 26 14:19:00 2024
Dec 26 19:09:05 DietPi radiod[6475]: Copyright 2023, Phil Karn, KA9Q. May be used under the terms of the GNU Public License
Dec 26 19:09:05 DietPi radiod[6475]: Loading config file /etc/radio/radiod@airspymini-autorx.conf...
Dec 26 19:09:05 DietPi radiod[6475]: Discovered airspy device serial: 10a862dc358f4463
Dec 26 19:09:05 DietPi radiod[6475]: Selecting 10a862dc358f4463; to select another, add 'serial = ' to config file
Dec 26 19:09:05 DietPi radiod[6475]: Airspy serial 10a862dc358f4463, hw version AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19, library version 1.0.11
Dec 26 19:09:05 DietPi radiod[6475]: 2 sample rates: 12000000 6000000
Dec 26 19:09:05 DietPi radiod[6475]: Set sample rate 12000000 Hz, offset 3000000 Hz
Dec 26 19:09:05 DietPi radiod[6475]: auto_rx: Software AGC 1; linearity 0, LNA AGC 0, Mix AGC 0, LNA gain 14, Mix gain 12, VGA gain 13, gainstep 21, bias tee 0
Dec 26 19:09:05 DietPi radiod[6475]: AGC thresholds: high -10.0 dBFS, low -40.0 dBFS
Dec 26 19:09:05 DietPi radiod[6475]: Locked tuner frequency 405600000.000 Hz
Dec 26 19:09:05 DietPi radiod[6475]: fftwf_import_system_wisdom() succeeded
Dec 26 19:09:05 DietPi radiod[6475]: fftwf_import_wisdom_from_filename(/var/lib/ka9q-radio/wisdom) succeeded
Dec 26 19:09:05 DietPi radiod[6475]: Processing [telemetry]
Dec 26 19:09:05 DietPi radiod[6475]: [telemetry] 1 channels started
Dec 26 19:09:05 DietPi radiod[6475]: Processing [manual-400]
[... 34 lines cut ...]
Dec 26 19:36:19 DietPi radiod[7031]: Airspy serial 10a862dc358f4463, hw version AirSpy MINI v1.0.0-rc10-0-g946184a 2016-09-19, library version 1.0.11
Dec 26 19:36:19 DietPi radiod[7031]: 2 sample rates: 12000000 6000000
Dec 26 19:36:19 DietPi radiod[7031]: Set sample rate 12000000 Hz, offset 3000000 Hz
Dec 26 19:36:19 DietPi radiod[7031]: auto_rx: Software AGC 1; linearity 0, LNA AGC 0, Mix AGC 0, LNA gain 14, Mix gain 12, VGA gain 13, gainstep 21, bias tee 0
Dec 26 19:36:19 DietPi radiod[7031]: AGC thresholds: high -10.0 dBFS, low -40.0 dBFS
Dec 26 19:36:20 DietPi radiod[7031]: Locked tuner frequency 405600000.000 Hz
Dec 26 19:36:20 DietPi radiod[7031]: fftwf_import_system_wisdom() succeeded
Dec 26 19:36:20 DietPi radiod[7031]: fftwf_import_wisdom_from_filename(/var/lib/ka9q-radio/wisdom) succeeded
Dec 26 19:36:20 DietPi radiod[7031]: Processing [telemetry]
Dec 26 19:36:20 DietPi radiod[7031]: [telemetry] 1 channels started
Dec 26 19:36:20 DietPi radiod[7031]: Processing [manual-400]
Dec 26 19:36:20 DietPi radiod[7031]: [manual-400] 0 channels started
Dec 26 19:36:20 DietPi radiod[7031]: 1 total demodulators started
Dec 26 19:36:20 DietPi radiod[7031]: airspy running
Dec 26 19:36:20 DietPi radiod[7034]: Established under name 'sonde-pcm.local'
Dec 26 19:36:20 DietPi radiod[7033]: Established under name 'airspymini-autorx'
Dec 26 19:36:20 DietPi radiod[7036]: Established under name 'sonde-pcm.local'
Dec 26 19:36:20 DietPi radiod[7035]: Established under name 'airspymini-autorx'
Dec 26 19:36:20 DietPi radiod[7043]: Established under name 'sonde.local'
Dec 26 19:36:20 DietPi radiod[7042]: Established under name 'auto_rx'
Dec 26 19:36:20 DietPi radiod[7047]: Established under name 'manual-400'
Dec 26 19:36:20 DietPi radiod[7045]: Established under name 'sonde-pcm.local'
Dec 26 19:36:20 DietPi radiod[7044]: Established under name 'telemetry'
Dec 26 19:36:20 DietPi radiod[7048]: Established under name 'sonde-pcm.local'
```

ke5gdb, 2024-12-26:
This is good news: 
```
Dec 26 19:36:20 DietPi radiod[7043]: Established under name 'sonde.local'
```
And it means that you likely need to wait an extra few seconds before starting auto_rx. Try starting KA9Q-Radio, waiting for that particular line item, and then starting auto_rx.

Feuerwerko, 2024-12-26:
Thats what I thought, but no matter how long I wait I still get the error. It establishes sonde.local but it still cant be reached by anything.

darksidelemm, 2024-12-26:
Depending on your distribution you might be having multicast issues. We know it works on Raspbian, but I'm not sure anyone has tried it on DietPi yet. There's some notes on these issues here: https://github.com/ka9q/ka9q-radio/blob/main/docs/notes.md#multicast-dns-problems-on-ubuntu

Feuerwerko, 2024-12-26:
Appears I have found a fix!
I had to install this package: libnss-mdns
[image]
Everything seems to work now!

closed 2024-12-26 by Feuerwerko

darksidelemm, 2024-12-26:
So it sounds like DietPi might be a little to 'diet' for this to work without extra packages...
Good to know.