projecthorus/radiosonde_auto_rx #968: lsusb parsing error: unexpected indentation

sgtkt, 2024-12-27:
I'm struggling with decoding of sonde data, which brought me to the debug log, where I noticed strange messages like:
```
DEBUG:lsusb parsing error: unexpected indentation: "   Port 1: 0000.0503 highspeed power enable connect"
DEBUG:lsusb parsing error: unexpected indentation: "   Port 2: 0000.0100 power"
DEBUG:lsusb parsing error: unexpected indentation: "   Port 3: 0000.0100 power"
DEBUG:lsusb parsing error: unexpected indentation: "   Port 4: 0000.0100 power"
DEBUG:lsusb parsing error: unexpected indentation: "   Port 5: 0000.0503 highspeed power enable connect"
DEBUG:lsusb parsing error: unexpected indentation: "   Port 1: 0000.0503 highspeed power enable connect"
```

These messages usually appear after these ones:
* `INFO:Sondehub Uploader - Uploaded station information to Sondehub.`
* `INFO:Task Manager - SDR #0 has been allocated to Scanner.`
* `INFO:Task Manager - SDR #0 has been allocated to Decoder (...).`

I've checked the source code of 'auto_rx' and 'lsusb', and it looks like the former expects 2 spaces as a hierarchy step, whereas the latter has 1 and 3 spaces in one specific section. Here's an excerpt from the output of 'lsusb -v' executed within the container (I've replaced all spaces with `·` to make it clear):

```
··bHubContrCurrent······1·milli·Ampere
··DeviceRemovable····0x02
··PortPwrCtrlMask····0xff
·Hub·Port·Status:
···Port·1:·0000.0503·highspeed·power·enable·connect
···Port·2:·0000.0100·power
···Port·3:·0000.0100·power
···Port·4:·0000.0100·power
···Port·5:·0000.0503·highspeed·power·enable·connect
Device·Qualifier·(for·other·device·speed):
··bLength················10
··bDescriptorType·········6
```

I'd appreciate if someone else could check for such messages in their logs.

HW: Raspberry Pi 3B + RTL SDR v3 Blog
OS: Raspbian: Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64
SW: radiosonde_auto_rx 1.8.0, docker version

darksidelemm, 2024-12-27:
These can be safely ignored, they are presented at the DEBUG level for a reason... 
The lines that we care about are decoded correctly. I should probably remove these debug lines.

argilo, 2025-02-09:
@darksidelemm Parsing lsusb's output sounds fragile. Is there a reason it's necessary, or could something like pyusb be used instead?

darksidelemm, 2025-02-10:
I'm sure it's possible. The main aim for using lsusb is to find the usb bus and device number for a particular RTLSDR serial number, so a USB reset can be applied. If the same information can be found using pyusb (and it looks like it probably can), then I'm sure it would be an option.

However, I haven't really had any drive to do it, since apart from these indentation errors on lines that aren't applicable to us, the lsusb approach has been working.

argilo, 2025-02-10:
I agree that if it ain't broke, don't fix it. 😄 PyUSB may be an option for the future in case the lsusb approach becomes problematic.

closed 2025-03-08 by darksidelemm