projecthorus/radiosonde_auto_rx #969: https://github.com/projecthorus/radiosonde_auto_rx/wiki/Airspy-Mini-Heatsink-Mods gone?

MikeTango, 2024-12-29:
Hi,

i am trying to make heads and tails of possibly conflicting information. 

https://github.com/projecthorus/radiosonde_auto_rx/wiki/Airspy-Mini-vs-R2-Notes states that you can have performance issues when setting Airspy Mini to 20mhz sample size, with a higher noisefloor.

The config that i am running (1.8.0) has a warning that the radio overheats when doing so and needs additional heatsinking

# Uncomment this to run your Airspy Mini at 20 MHz sample rate (same at the Airspy R2)
# NOTE - This can result in your Airspy Mini overheating if you do not add additional thermal material inside it, and heatsink the case!
#samprate = 20000000

The corresponding page https://github.com/projecthorus/radiosonde_auto_rx/wiki/Airspy-Mini-Heatsink-Mods but is gone. 

I found this change from two weeks ago

https://github.com/projecthorus/radiosonde_auto_rx/wiki/KA9Q%E2%80%90Radio-Setup-Notes/_compare/af97464e1d1436f8d169eb3d4fcfdfabe01e0de8...828f7fa94a532cc43fe9b7cb6dc075f6e8bb6ee9

TL;DR is the heat problem a non issue now?

darksidelemm, 2024-12-29:
It seems that heat isn't the issue, correct. Spurs are the issue (if they are an issue at all)

MikeTango, 2024-12-30:
Would you mind adding the disassembly instructions again just for info? I am running the radio in an enclosed box and every bit counts.

darksidelemm, 2024-12-30:
Doesn't look like I can revert the edits, sorry.... The best I can do is point you to these photos - https://photos.app.goo.gl/H815go2PG5B9LXm38

MikeTango, 2024-12-31:
[image]
[image]

just took out the old trusty FLIR

darksidelemm, 2024-12-31:
Sure...  but the case style makes it kind of difficult to just bond one small IC to the case. If you just use a small bit of thermal material, it's likely to get pulled off the board as you slide it into the case again.

MikeTango, 2025-01-01:
I used some stuff that came with a raspi, there is a small copper piece that i put on top of the stacks, it slides in easy. The pressure on the big IC side pushes the board towards the other side of the casing and the chip on that side now has direct contact with the case. I admit that it's a bit tricky to slide in like that. For the external cooling fins i used some NVME2 format radiators that i had lying around. I tried active cooling but the fan motor plugged into the raspi GPIO introduced all kinds of interference. 
[image]

This system is my portable, it's deployed in a marine environment and hasn't made any problems so far.

MikeTango, 2025-01-02:
[image]

This is from my permanent setup, no issues so far

darksidelemm, 2025-01-02:
Oh the station with all the emojis that I get the occasional complaint about it YOU!

Keep it up :-)

Glad to hear it works!

closed 2025-01-02 by darksidelemm

MikeTango, 2025-01-02:
It is absolutely not my fault that UTF-8 exists.

[image]