projecthorus/radiosonde_auto_rx #970: Pushbullet support?

l0r-e, 2024-12-31:
Hello, i was wondering if there was any possibility to have more alerts directly from auto_rx, instead of classic emails, made for those not so much terminal familiar. 

I think their api's easy to work with. I already tried by adding a kind of alerting script (just for ascending sondes) but nothing, i am far too unexperienced to do it myself.

darksidelemm, 2024-12-31:
I don't want to add more alerting service integrations into auto_rx, sorry.

auto_rx already emits all telemetry it receives via UDP broadcast. 
If someone wants a particular alerting system, then I recommend they make use of this output and write their own alerting software based on it.

closed 2024-12-31 by darksidelemm

l0r-e, 2024-12-31:
oh, okay. fantastic.
Looks like i missed that part! thanks for the quick response :)

buttim, 2024-12-31:
Someone already built something to do what you want I think. Not tested personally
https://github.com/ch3p4ll3/radiosonde_auto_rx_notifier

l0r-e, 2025-01-01:
Oh cool, thanks

darksidelemm, 2025-01-01:
That's a neat project! I think i need to start a list of 'auto_rx add-on software' like this, so they can be more easily found...