projecthorus/radiosonde_auto_rx #975: KA9Q offset issues

MikeTango, 2025-01-05:
Hi,

i have some issues understanding the upper frequency offset on the KA9Q radio settings. 
In the instruction it says to offset it by 600khz above the max target frequency. 
When setting the radiod config to 
`frequency = 406m6 `
and the station.cfg to 
`min_freq = 400.00
max_freq = 406.00` 
this results in the following spectrum image:
[image]
With a noticable trough when nearing the end. The main spike of the resonator frequency is not at 406,6 but at 406.0. 
Without changing the station.cfg bracket, offsetting the frequency up by .1 to
`frequency = 406m7`
shows the same spectrum after a reboot:
[image]
Again, offsetting the frequency up by .1 to
`frequency = 406m8`
[image]
Up by .1 to
`frequency = 406m9`, 
a wild radiosonde appears and messes with the layout, cheeky bugger, but the spike doesn't move:
[image]
And then, at
`frequency = 407m0`: 
Pop, it's gone. 
[image]

When setting it to lower values like
`frequency = 405m5`: 
it does the same, it "sticks" to the 405 mark and only jumps at 1mhz steps:
[image]

Is there any way to explain this behaviour? Does it affect the reception in the areas of the trough? 

Regards,

MT

darksidelemm, 2025-01-05:
Is this with an Airspy Mini or R2? What have you got the samprate setting set to in your ka9q-radio config?

MikeTango, 2025-01-06:
Hi
forgot to add. This is after setting the samplerate to 20mhz on the Airspy Mini.

MikeTango, 2025-01-06:
This ist with commented 
'#samprate = 20000000'
'frequency = 405m6'

[image]

Also commented samplerate, but 900khz up, you can see a drop off at the low end as expected:
'#samprate = 20000000'
'frequency = 406m5'
[image]

darksidelemm, 2025-01-06:
OK... so I'm not really sure what you are considering the 'expected behaviour' to be here.

Some points:
- KA9Q tunes the Airspy in 'Real' mode, where the available (initial) spectrum is *below* the frequency you define in the radiod configuration file. By default KA9Q runs the Airspy mini at 12 MHz real sample rate (for <~5.5 MHz usable bandwidth), and the Airspy R2 at 20 MHz real sample rate (for <~8 MHz usable bandwidth). 
- When running an Airspy Mini at 20 MHz real sample rate, there are additional spurs present, due to the 'poorer' reference clock. These spurs are not fixed in absolute frequency, and will instead be seen at fixed offsets from the SDRs tuned frequency. This is what 
- If you define a frequency min/max range within auto_rx that exceeds the initial available spectrum range from a ka9q-radio server, then that server is going to re-tune to cover that range when auto_rx requests a scan plot. This may result in undesirable behaviour, hence we give particular recommended configurations to use.  I suspect the 'troughs' you are seeing are a result of the ka9q-radio server re-tuning, resulting in the edges of the filter skirts ending up somewhere bad.

MikeTango, 2025-01-06:
The recommended settings exhibit the shown behavior with radiod at

'frequency = 406m6 '
'samprate = 20000000'

'min_freq = 400.00 max_freq = 406.00'

[image]

So the edges of the filter skirts are somewhere bad when using the recommended settings. The plot shows a noticeable dropoff at the upper end, which to me means that this part is unusable. Trying to adjust that, I expect that all artefacts move in .1 increments when offsetting the frequency by .1, including the filter skirts, but no matter what, they jump in 1Mhz steps on the nearest lower frequency without decimals.

 So it feels like the config is ignoring the decimals when setting the initial upper frequency?

darksidelemm, 2025-01-06:
Actually... for 20 Msps, you should probably be going with the recommended Airspy R2 settings, which has frequency set to 407 MHz... so I'd suggest changing to this. 

It's quite possible that there is something not quite right happening with the centre frequency in ka9q-radio, but that's going to require more debugging with ka9q-radio.

MikeTango, 2025-01-06:
Done, but the behaviour is also showing with a sample rate of 12. Should i raise the issue in their git repo?

ka9q, 2025-01-30:
Hi, KA9Q here. What you're seeing in those plots is totally normal. The Airspy R2/mini uses "high side" injection with a post-mixer bandpass filter that runs from about -9.4 MHz up to -0.6 MHz relative to the tuned "carrier" frequency. This is what the A/D converter samples, and KA9Q-radio slices it up into whatever pieces you want.

No superheterodyne (LO/mixer type) receiver can properly respond all the way up (or down) to its tuner frequency; if you try you'll get part of the unwanted sideband aliasing into your output, and once that happens you can't get rid of it. Also the output gets rather noisy as you approach the local oscillator frequency because of its inherent phase noise, which goes as 1/f or 1f^2 or something like that. And there's a big "spike" right at it where the receiver sees its own local oscillator. This can be notched out but the noise cannot, so it's best to stay away from it altogether.

You can see this low-frequency noise with SDRs that have two A/D hardware converters natively producing complex (I/Q) output, e.g, the SDRPlay and the Funcube dongle. (Internally the Airspy R2 has just one 20 MHz A/D converter running on the tuner output. The _driver library_ normally converts that to 10 MHz complex but the implementation is inefficient and, in the case of KA9Q-radio, unnecessary. I just ask for the raw real-only stream since I can process it with a real-to-complex FFT at zero additional cost. KA9Q-radio handles both types of front end by just using the appropriate FFTs.

Hence the analog filter in the Airspy that rolls off below 600 kHz, giving it room to reach zero before it gets to DC. There's no such thing as a "brickwall" filter, especially in analog.

Remember all this looks upside down in the spectrum display because the front end uses high side injection, i.e., you can think of it as a "lower sideband" receiver with a -9.4 MHz to -600 kHz "sideband".

The Airspy, Funcube, SDRPlay, RTL-SDR and other "tunable" SDR front ends use a tuner chip originally designed for digital TV, hence the 8+ MHz bandwidth. (The USA uses 6 MHz channels; some countries use 8 MHz.)

Managing the local oscillator (LO) in a multi-channel receiver like KA9Q-radio is a challenge because it might not be able to satisfy everyone simultaneously. A newly created or manually tuned KA9Q-radio channel will, if necessary, retune the front end to include its requested frequency range but to prevent a "tuning war" the other channels will just go silent if they're no longer in the tuner's pass band. It's easy to do this by accident and cut off a whole bunch of important channels so I provide the "frequency" directive in the hardware section of the config file to lock the tuner LO to a specified frequency. Setting this requires knowledge of the front end, specifically the response of its filter and whether it uses high or low side injection. Hence the recommendation to lock the Airspy tuner at 406.6 MHz or so. (I'm intrigued by the idea of tuning it a little higher to also cover the EPIRB band at 406.0-406.1 MHz, though someone would have to develop a decoder).

Whatever the front end frequency, KA9Q-radio automatically sets its "second LO" (the one in software) in response to give the requested RF channel frequency. You can manually move the LO with my _control_ program; this is useful for moving spurs away from important signals.

The ideal SDR receiver uses "direct sampling", where the antenna goes directly to a (very fast) A/D converter. It still needs an analog "anti alias" filter to remove everything above one half of the sample rate, but there is no local oscillator. (Think of the local oscillator being fixed at 0 Hz.) Except for rolloff in RF transformers and DC blocking capacitors, these receivers can theoretically go all the way down to DC on the antenna input. These are now becoming popular on HF all the way through 6m, but beyond that is still out of reach on a ham budget. Hence the use of TV-type front ends for VHF and above. Even the RX888, a direct sampling front end that can clock at 130 MHz, uses a TV-type tuner to reach frequencies above 6m. Radiosondes are at 400-406 MHz, so frequency conversion is still common practice.

There is an intriguing way for a direct sampling receiver to operate on higher frequencies through _controlled_ use of aliasing. Put an analog band pass filter on the input to select a multiple of the sample rate and everything will be shifted down by that multiple. This will work up to the frequency limit of the sample-and-hold inside the A/D converter; for the 2208 in the RX888, this is 750 MHz. (KA9Q-radio already supports this, but it hasn't actually been tested.) Of course, any hardware low pass filter in the receiver must be removed.

argilo, 2025-02-06:
I tracked down the root cause in ka9q-radio. Details here: https://github.com/ka9q/ka9q-radio/issues/101#issuecomment-2639975714

argilo, 2025-02-07:
Phil pushed a fix for the bug, so we should test out a newer commit and recommend it in the wiki.

argilo, 2025-02-07:
I updated my stations to 21f7f9a70e619ca88394ff1ca86de7482340232f and will report back on how that goes.

MikeTango, 2025-02-07:
Running on that commit, looks good so far.

closed 2025-03-01 by MikeTango