projecthorus/radiosonde_auto_rx #977: Multiple RTL-SDR problem - solved ???

SQ3H, 2025-01-16:
Hi all. 
Sorry sorry my english

I think I found the solution Multiple RTL-SDR problem.
While trying to upgrade the receiver I encountered a known bug with more than 3 rtl-sdr dongles - "rtl_sdr test call resulted in return code of 5".
Changing serial numbers, changing the order of dongles, and separating dongles to different USB hubs did not work.
I have eliminated the possibility of low USB bandwidth, USB power issues and other technical issues.
Przeglądając fora i grupy dyskusyjne natrafiłem na ciekawą koncepcję - numbering dongles with a geometric sequence n=2*n (2, 4, 8, 16, 32 etc.)
The first tests showed that it could work.

Hardware: Dell Wyse 5060, AMD GX-424cc), 16GB RAM, 128GB SSD, 9 pcs of RTL-SDR, USB hub, other necessary stuff.
Software: Debian 12, radiosonde_auto_rx 

Additionally horusdemodlib and rtlsdr-wsprd are running on the same machine on additional rtl's.

The auto_rx dongles were named as follows:
RTL008, RTL016, RTL032, RTL064, RTL128, RTL256 and RTL512 (yes, 7 pcs ! ) .

First attempt - repeated multiple times manual decoding forced - passed!
Combat test - passed!!!

The whole thing seems to work stably. rtl_sdr does not respond with any errors.

I am still testing this configuration
Please conduct tests on other machines.
Please conduct tests with other numbering sequences (n=3*n etc.)

[image]
[image]
[image]
[image]
[image]

SQ3H, 2025-01-16:
11.00UTC - retest passed. autorx uptime 13h, second series of decoding on all receivers.

[image]

ke5gdb, 2025-01-16:
Did the previous serial numbers include any letters? Some of my problems were solved when I changed the serial numbers from "001", "002", .. to "SN001", "SN002", ...

SQ3H, 2025-01-16:
> Did the previous serial numbers include any letters? Some of my problems were solved when I changed the serial numbers from "001", "002", .. to "SN001", "SN002", ...

Yes, I have tried different numbering methodologies.

For example:
- 00000001 to 00000006
- 00000101 to 00000106
- 00000101 to 00000103, 00000201 to 00000203
- RTL001 to RTL006
- RTL101 to RTL106
etc.

Each time the 4th receiver went into the decoding process, an error occurred - "rtl_sdr test call resulted in return code of 5", after which the receiver was removed from the list in auto_rx (it was still present in rtl_test)

Assigning serial numbers according to the scheme RTL008, RTL016, RTL032, RTL064..... solved the problem with "rtl_sdr test call resulted in return code of 5".
rtl_sdr seems to work correctly with such numbering.

> Did the previous serial numbers include any letters? Some of my problems were solved when I changed the serial numbers from "001", "002", .. to "SN001", "SN002", ...

darksidelemm, 2025-01-16:
It's good that you've had some success, but once you're at more than 2 RTLSDRs, it's *really* time to move to an Airspy-based setup...

SQ3H, 2025-01-16:
> It's good that you've had some success, but once you're at more than 2 RTLSDRs, it's _really_ time to move to an Airspy-based setup...
> 
> I should also note that your solution might work on your platform, but there are also likely other issues causing RTLSDR failures on other platforms (e.g Raspberry Pi), such as USB bus bandwidth.

Yes, switching to Aairspay may have been justified. But I happen to have a few redundant rtl-sdrs that I can and want to use.

Nevertheless, it is about solving a problem that has remained unsolved in the auto_rx user community for several years. If only to get it over with ;)

I am aware that the method used may not work on other platforms. That is why I asked for a test if possible

argilo, 2025-02-06:
I did some experimenting with `rtl_test`, and found that when more than four RTL-SDRs are used simultaneously, the following message appears:

```
Failed to submit transfer 3
Please increase your allowed usbfs buffer size with the following command:
echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb

Library error -5, exiting...
```

After following the suggestion to increase the USB memory limit with `sudo sh -c 'echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb'` I was able to run many more RTL-SDRs simultaneously, including in Auto-RX.

When testing on a Raspberry Pi, I also had to use powered USB hubs, since running more than four RTL-SDRs simultaneously would cause an over-current condition, resetting all USB devices.

> Some of my problems were solved when I changed the serial numbers from "001", "002", .. to "SN001", "SN002", ...

The issue there is that if the device argument can be parsed as an integer and is less than the number of connected devices, then it is interpreted as a device index instead of a serial number.

https://gitea.osmocom.org/sdr/rtl-sdr/src/commit/619ac3186ea0ffc092615e1f59f7397e5e6f668c/src/convenience/convenience.c#L264-L270

SQ3H, 2025-02-09:
For information: auto_rx with named RTL-SDR using my method has an uptime of over 3 weeks and works without errors all the time.

argilo, 2025-02-09:
Are you sure that nothing else changed in your software environment? I had a look over how the RTL-SDR driver and Auto-RX process serial numbers, and I haven't found any reason why the numbering scheme should matter (apart from the need to avoid fully-numeric serial numbers that are lower than the number of connected devices).

SQ3H, 2025-02-09:
Nothing else has changed.

Note 1: increasing the "usbfs_memory_mb" buffer before changing serial numbers did not cause more rtl-sdr's to work correctly.

Note 2: Note that auto_rx works with 7 rtl-sdr's, and the lowest one hat number RTL008. I haven't checked this, but it's possible that you're absolutely right, and I accidentally named the first receiver with a number higher than the number of connected receivers.

argilo, 2025-02-10:
> it's possible that you're absolutely right, and I accidentally named the first receiver with a number higher than the number of connected receivers

This situation would only occur if you specified a completely numeric serial number in your Auto-RX configuration (e.g. `3` or `00000003`). If letters are included (e.g. `RTL003`) then the device search should work correctly.

One further observation about serial number matching in the RTL-SDR utilities: If the specified argument was not an integer lower than the number of connected devices, and did not match the serial number of any connected device, then "prefix" and "suffix" searches will be attempted. That is, the program will first look for a device whose serial number begins with the specified value, and if it finds none it will look for a device whose serial number ends with the specified value:

https://gitea.osmocom.org/sdr/rtl-sdr/src/commit/619ac3186ea0ffc092615e1f59f7397e5e6f668c/src/convenience/convenience.c#L281-L303

I suppose the prefix & suffix searches could also lead to the wrong device being selected in certain situations.

argilo, 2025-02-10:
That said, attempting to access an RTL-SDR that is already in use seems to result in return code 1, not return code 5. I see return code 5 when there is insufficient USB memory. (It's possible that there are other scenarios which would trigger return code 5, though.)