projecthorus/radiosonde_auto_rx #990: Using KA9Q with RTL-SDR

michaelselfwx, 2025-02-26:
Is there any possible way to use a rtl-sdr (or 2) with the KA9Q and auto_rx?

DL2MF, 2025-03-02:
Yes, simple use ist, it works very fine ;-)

darksidelemm, 2025-03-02:
While this is possible, it's not a use-case that we support, or will provide documentation on.
ka9q-radio fixes the centre frequency of the SDR, so a RTLSDR will only ever be able to cover ~2 MHz of bandwidth, which is insufficient to cover the entire radiosonde band. It *may* be usable in some locations, but in general, it's not suitable.

Using *multiple* RTLSDRs is definitely not supported. This would require running 2x ka9q-radio servers, and significant changes to auto_rx. This is not something that is planned at this time. Buy an Airspy Mini instead.

closed 2025-03-02 by darksidelemm