projecthorus/radiosonde_auto_rx #992: Docker launching issue (not due to projecthorus)

f4gbv, 2025-02-26:
hello, 

issue : "docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock"
(I'm installing projecthorus on a rpi pi zéro v.1.3 )

solved here : 

"[on Nov 12, 2020](https://github.com/Xilinx/Vitis-AI/issues/18#issuecomment-726156212)

sudo chmod 666 /var/run/docker.sock

"

73 isa f4gbv:)

darksidelemm, 2025-02-26:
This suggests that you haven't added your user to the docker group.

closed 2025-02-26 by darksidelemm

f4gbv, 2025-02-26:
> This suggests that you haven't added your user to the docker group.

hello,  I  really did all steps  as asked  :  "sudo usermod -aG docker $(whoami)" 

As this error occured, I did it again, but it did'nt change anything.

Then next,  I  did : "sudo chmod 666 /var/run/docker.sock"
and  finally it worked

darksidelemm, 2025-02-26:
For the usermod change to take effect, you have to at least log out and log back in again, though restarting is usually safer.