projecthorus/radiosonde_auto_rx #993: log issue

MikeTango, 2025-03-01:
Moin,

today i chased U4254142 and while the track gets displayed until i lost contact, the log only has the ascent until 11000m saved. The data must be somewhere, else it would not be on the map.

[image]

[autorx_logfiles_❤️🧡💛💚💙💜 Dreamcatcher 💜💙💚💛🧡❤️_20250301-152715Z.zip](https://github.com/user-attachments/files/19039496/autorx_logfiles_.Dreamcatcher._20250301-152715Z.zip)

I just pulled the data for another sonde with lots of frames, and it's the same thing, the plot is complete but the log ends after not even half the flight.

Where's the rest of the log?

DL2MF, 2025-03-02:
Meteorological skew plots are usually limited to the 100 hPa level, also in professional applications, to be comparable.

darksidelemm, 2025-03-02:
OK, I see you updated the issue...
Yes, the skewt plot stops at 100 hPa. That's a hardcoded thing in the SkewT plot generation. There is generally not much need to go higher than this.

darksidelemm, 2025-03-08:
I'm not sure there's much more to say on this one? The skewT plot stops at 100 hPa - this is expected behaviour.

closed 2025-03-08 by darksidelemm