projecthorus/radiosonde_auto_rx #1001: Feature Request - Store decoded data in RDBMS

AlNotSteve, 2025-04-14:
Hi,

Would you consider the ability to send data directly to a relational database such as PostgreSQL?

darksidelemm, 2025-04-14:
In short, no. We emit the live data via UDP output, and the log files on disk. 

If you want data to end up in some other database, then I suggest capturing the data from one of these outputs and feeding it into the database you want.

closed 2025-04-14 by darksidelemm

AlNotSteve, 2025-04-25:
Thanks.  Was venturing down that path, but wanted to check if it was on your roadmap before I proceeded.
I've got all my data going into PostgreSQL by ingesting the cleaned up CSV files.  I've removed all entries with invalid temperatures and humidity readings.  Appears the sensors only activate a short while after I start receiving data.