projecthorus/radiosonde_auto_rx #1003: utils/plot_sonde_log.py not plotting data above 9km

AlNotSteve, 2025-04-16:
Hi,

Started to use utils/plot_sonde_log.py and see that the output PNG files only go up to 9km.

Example below.

[image]

darksidelemm, 2025-04-16:
Looks like the altitude limit is set to a pressure level of 300 hPa here:
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/plot_sonde_log.py#L330

Change that to a lower pressure if you want to plot higher.

AlNotSteve, 2025-04-16:
Thanks for that.  Looks better, however the temperature scale is not right.

[image]

darksidelemm, 2025-04-16:
Which will be set here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/plot_sonde_log.py#L334

Looks like I changed the code to use metpy and didn't pass through any of the command line arguments. I'll try and fix it when i have the time.

AlNotSteve, 2025-04-16:
Cool.

No good after changing to "skew.ax.set_xlim(-70, 30)"

[image]

AlNotSteve, 2025-04-16:
Noticed the scale issue also affects the plots in the web UI

darksidelemm, 2025-04-16:
I think you're mis-interpreting the plot. 
This is a Skew-T plot, where the temperature lines are 'angled' or 'skewed' across the plot. 

Refer here for more information: https://flsc.org/portals/12/PDF/Read_Skew_T.pdf

AlNotSteve, 2025-04-17:
Cool, thanks for the doc. Just assumed the temperature lines would match to the temperature scale horizontally. Will enjoy the read over the weekend 😁

AlNotSteve, 2025-04-17:
Had a bit of a read. Clears it up nicely. Thanks for that!

AlNotSteve, 2025-04-25:
Learned a lot about Skew-T graphs.  Awesome stuff in them.
I've made some changes to the script on my side and aligned them more with BOM soundings.
I have made my soundings for Brisbane available to those of you that are interested.  Go to https://rs.inoz.au/soundings/.  I will keep them for as long as I have disk space available.
Do you want me to do a pull request with my changes to contribute back?

closed 2025-04-25 by AlNotSteve