projecthorus/radiosonde_auto_rx #1004: Filter Telemetry in Real-Time

michaelselfwx, 2025-04-30:
It seems to be the case that almost every DFM-17 sonde that I track occasionally has one frame that has a completely wrong position. Sometime this can be just 1 more degree latitude/longitude or sometimes it is something like 5 degrees. Either way, it is annoying because it messes with my coverage maps, maximum distances (on radiosondy.info), and makes the UI look wonky.

I would love for there to be a way to filter out these data points that are unrealistically far away from the last data point. I would think the best way to do this is to make sure the distance/time is not more than like 500 kph or something along those lines.

DL2MF, 2025-05-02:
> I agree this would be useful, but its going to take a bit of thought to do it in a way that won't cause other issues.
> 
> Initial thoughts:
> 
...
> 
>     * Something like a running average on lat/lon/alt might work, but even then you'd need special cases to handle cases like loss of signal, where on re-acquisition the position could jump significantly.

Full ack to all topics, just an idea concerning the last one:
- if the time delta is greater than n seconds the position may move (n * speed ) + spread value

I'm using a similar approach in my landing notification bot to avoid continously landing messages when a sonde is a treelanding in the RX coverage area.

darksidelemm, 2026-02-11:
Released in 1.8.2!