projecthorus/radiosonde_auto_rx #1005: M20 radiosonde temperature and humidity not decoding properly

tushroy, 2025-05-14:
M20 radiosonde temperature and humidity not decoding properly.
```
timestamp,serial,frame,lat,lon,alt,vel_v,vel_h,heading,temp,humidity,pressure,type,freq_mhz,snr,f_error_hz,sats,batt_v,burst_timer,aux_data
2025-05-14T05:35:01.000Z,M20-309-3-03709,1431236101,23.78259,90.37934,170.4,2.7,9.1,356.0,-273.0,-1.0,-1.0,0x20,402.466,24.2,-1875,-1,2.6,-1,-1
2025-05-14T05:35:02.000Z,M20-309-3-03709,1431236102,23.78266,90.37935,174.2,5.2,6.5,30.6,-273.0,-1.0,-1.0,0x20,402.466,24.2,-1875,-1,2.6,-1,-1
```
always getting values like -273.0, -1 

[20250514-053443_M20-309-3-03709_M20_402466_sonde.log](https://github.com/user-attachments/files/20205503/20250514-053443_M20-309-3-03709_M20_402466_sonde.log)

darksidelemm, 2025-05-14:
Hmm. Unsure if something has changed in the telemetry, a different model, or if perhaps this is actually a PS-20. Do you actually have the sonde in-hand?

Can you enable the save_raw_hex option here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L524
This will result in some .raw files in your log directory created during each reception. Upload these somewhere for us to have a look at.

also paging @rs1729

tushroy, 2025-05-14:
@darksidelemm Hi Mark, I don't have the sonde in hand. I have enabled the `save_raw_hex` option as you suggested. Hopefully, I will receive a new signal in approximately 3 hours. I have also enabled `save_detection_audio` and `save_decode_iq` to better understand the issue.

My local weather forecasting authority uses two types of sondes. The other one is detected as IMET4. You may be right about the PS-20. I will upload and share the log files once I receive a transmission detected as M20.

tushroy, 2025-05-15:
@darksidelemm Hi Mark, i have uploaded `M20_401467_sonde.log` and `M20_401467000.raw` (renamed the `raw` to `txt` for uploading). Please have a look.

[20250514-173936_M20-310-3-01971_M20_401467_sonde.log](https://github.com/user-attachments/files/20217991/20250514-173936_M20-310-3-01971_M20_401467_sonde.log)

[20250514-173933_M20_401467000.raw.txt](https://github.com/user-attachments/files/20217995/20250514-173933_M20_401467000.raw.txt)

rs1729, 2025-05-15:
Serial number is "310-3-...." from 2023, so no new model I guess.
There is T-ADC data, but the calculated value is above +60C, thus no output (T=-273C).
Around T=64C all the time (output at the end of line). And the RH-ADC constant at 0xbb80.
Back 2023 there was a PS20 showing such values, that's why the upper limit of T=60C was introduced. There was not much more to identify PS20. Don't have many PS20 samples though.
```
Wed 2025-05-14 17:39:54.000  lat: 23.80602  lon: 90.38711  alt: 948.39   vH:  7.6  D:  27.6  vV: 3.4   SN: 310-3-01971  #  [OK]  T:64.4C TH:26.3C
Wed 2025-05-14 17:39:55.000  lat: 23.80610  lon: 90.38715  alt: 951.35   vH: 11.5  D:  23.0  vV: 2.9   SN: 310-3-01971  #  [OK]  T:64.4C TH:26.4C
[...]
Wed 2025-05-14 18:00:59.000  lat: 23.79884  lon: 90.47185  alt: 5200.77   vH: 13.6  D: 116.8  vV: 4.1   SN: 310-3-01971  #  [OK]  T:64.7C TH:0.2C
Wed 2025-05-14 18:01:10.000  lat: 23.79855  lon: 90.47307  alt: 5236.46   vH: 15.8  D: 115.1  vV: 2.5   SN: 310-3-01971  #  [OK]  T:64.5C TH:-0.1C
[...]
Wed 2025-05-14 18:34:28.000  lat: 23.73250  lon: 90.66721  alt: 12142.90   vH:  9.2  D: 102.3  vV: 2.0   SN: 310-3-01971  #  [OK]  T:64.4C TH:-43.9C
Wed 2025-05-14 18:34:31.000  lat: 23.73242  lon: 90.66740  alt: 12154.29   vH: 10.0  D:  99.0  vV: 3.5   SN: 310-3-01971  #  [OK]  T:64.4C TH:-43.9C
```
The TH temperature (humidity sensor) shows plausible values. So it seems to have at least this small sensor.
(Or does the PS20 use the MCU temperature in the output instead?)

tushroy, 2025-05-15:
@rs1729 got another short pass. (if that helps anyway)

[20250515-053458_M20_402467000.raw.txt](https://github.com/user-attachments/files/20224290/20250515-053458_M20_402467000.raw.txt)
[20250515-053502_M20-309-3-04004_M20_402467_sonde.log](https://github.com/user-attachments/files/20224289/20250515-053502_M20-309-3-04004_M20_402467_sonde.log)

tushroy, 2025-05-15:
Hi @rs1729 can you have a look on this : https://github.com/dbdexter-dev/sondedump/blob/528655564f34a42e464334fced4f7e89f9bfee46/sonde/m10/parser.c#L209

rs1729, 2025-05-15:
> [@rs1729](https://github.com/rs1729) got another short pass. (if that helps anyway) [...]

Looks very similar:
```
Thu 2025-05-15 05:35:19.000  lat: 23.77749  lon: 90.37438  alt: 450.23   vH:  9.4  D: 253.5  vV: 4.5   SN: 309-3-04004  #  [OK]  T:65.4C TH:27.2C
Thu 2025-05-15 05:35:24.000  lat: 23.77737  lon: 90.37400  alt: 473.77   vH: 11.0  D: 249.6  vV: 7.4   SN: 309-3-04004  #  [OK]  T:65.0C TH:27.1C
Thu 2025-05-15 05:35:25.000  lat: 23.77734  lon: 90.37390  alt: 479.07   vH: 10.3  D: 260.9  vV: 3.4   SN: 309-3-04004  #  [OK]  T:65.4C TH:27.1C
Thu 2025-05-15 05:35:27.000  lat: 23.77730  lon: 90.37375  alt: 487.17   vH:  8.5  D: 271.1  vV: 3.4   SN: 309-3-04004  #  [OK]  T:65.2C TH:27.2C
Thu 2025-05-15 05:35:35.000  lat: 23.77715  lon: 90.37317  alt: 520.67   vH:  4.1  D: 252.8  vV: 3.0   SN: 309-3-04004  #  [OK]  T:65.2C TH:27.0C
Thu 2025-05-15 05:36:01.000  lat: 23.77695  lon: 90.37142  alt: 616.16   vH:  6.0  D: 266.0  vV: 3.2   SN: 309-3-04004  #  [OK]  T:65.4C TH:26.9C
Thu 2025-05-15 05:38:59.000  lat: 23.78305  lon: 90.36841  alt: 1238.71   vH:  5.7  D:  20.1  vV: 4.1   SN: 309-3-04004  #  [OK]  T:65.6C TH:24.9C
Thu 2025-05-15 05:40:40.000  lat: 23.78792  lon: 90.36941  alt: 1622.86   vH:  8.3  D:   3.3  vV: 3.0   SN: 309-3-04004  #  [OK]  T:65.3C TH:22.6C
Thu 2025-05-15 05:41:18.000  lat: 23.78961  lon: 90.37002  alt: 1759.53   vH:  2.1  D: 343.8  vV: 2.8   SN: 309-3-04004  #  [OK]  T:65.8C TH:21.4C
```

rs1729, 2025-05-15:
> Hi [@rs1729](https://github.com/rs1729) can you have a look on this : https://github.com/dbdexter-dev/sondedump/blob/528655564f34a42e464334fced4f7e89f9bfee46/sonde/m10/parser.c#L209

Reminds me of
https://github.com/rs1729/RS/blob/master/demod/mod/mXXmod.c#L564
Does it give any other values?

For normal M20 with valid RH data, the temperature calculations seem to work fine.
Here the ADC-values are very low giving T>64C.

rs1729, 2025-05-15:
Are these normal radiosonde soundings at 00Z/12Z the same? These two recordings show 18Z and 06Z.
Or is there another radiosonde in the air at the same time? Who is launching this? Looks like the Department of Environment or Costguard, but I see other tech/science buildings on the map, and what seems to be a military airfield.

Maybe they launch imet for PTU and additional PS20 for wind?
Maybe you can track one down and recover.

tushroy, 2025-05-15:
@rs1729 can you check on following: 
```
static float get_Temp(gpx_t *gpx) {
    // Steinhart-Hart coefficients for Shibaura PB5-41E
    const float p0 = 1.07303516e-03;
    const float p1 = 2.41296733e-04;
    const float p2 = 2.26744154e-06;
    const float p3 = 6.52855181e-08;

    // Bias and parallel resistors
    const float Rs[3] = {12.1e3, 36.5e3, 475.0e3};
    const float Rp[3] = {FLT_MAX, 330.0e3, 2e6}; // FLT_MAX ≈ infinity

    const uint16_t adc_max = 4095;

    // Read raw ADC value (same format as m20_20_temp)
    uint16_t raw_val = (gpx->frame_bytes[0x5] << 8) | gpx->frame_bytes[0x4];
    uint16_t adc_val = raw_val & 0x0FFF;  // Lower 12 bits = ADC value
    uint8_t range_idx = raw_val >> 12;    // Upper 4 bits = range

    if (range_idx > 2) return 0;  // Invalid range index

    // Convert to ADC ratio (0.0–1.0)
    float adc_percent = adc_val / (float)adc_max;

    // Calculate thermistor resistance based on voltage divider
    float R;
    if (range_idx == 0) {
        R = adc_percent * Rs[0] / (1.0f - adc_percent);
    } else {
        float Rp_eff = Rp[range_idx];
        float Rs_val = Rs[range_idx];
        float denom = Rp_eff - adc_percent * (Rs_val + Rp_eff);
        if (fabs(denom) < 1e-6f) return 0;  // Avoid division by near-zero
        R = adc_percent * Rs_val * Rp_eff / denom;
    }

    // Calculate temperature using Steinhart-Hart equation
    if (R <= 0) return 0;

    float logR = logf(R);
    float T_kelvin = 1.0f / (p0 + p1 * logR + p2 * logR * logR + p3 * logR * logR * logR);
    float T_celsius = T_kelvin - 273.15f;

    // Sanity check
    if (T_celsius < -120.0f || T_celsius > 60.0f)
        return 0;

    return T_celsius;
}
```

tushroy, 2025-05-15:
> Reminds me of https://github.com/rs1729/RS/blob/master/demod/mod/mXXmod.c#L564 Does it give any other values?
>
> For normal M20 with valid RH data, the temperature calculations seem to work fine. Here [...]

@rs1729 can you please check on following: 

```
static float get_Temp(gpx_t *gpx) {
    // Steinhart-Hart coefficients for Shibaura PB5-41E
    const float p0 = 1.07303516e-03;
    const float p1 = 2.41296733e-04;
    const float p2 = 2.26744154e-06;
    const float p3 = 6.52855181e-08;

    // Bias and parallel resistors
    const float Rs[3] = {12.1e3, 36.5e3, 475.0e3};
    const float Rp[3] = {FLT_MAX, 330.0e3, 2e6}; // FLT_MAX ≈ infinity

    const uint16_t adc_max = 4095;

    // Read raw ADC value (same format as m20_20_temp)
    uint16_t raw_val = (gpx->frame_bytes[0x5] << 8) | gpx->frame_bytes[0x4];
    uint16_t adc_val = raw_val & 0x0FFF;  // Lower 12 bits = ADC value
    uint8_t range_idx = raw_val >> 12;    // Upper 4 bits = range

    if (range_idx > 2) return 0;  // Invalid range index

    // Convert to ADC ratio (0.0–1.0)
    float adc_percent = adc_val / (float)adc_max;

    // Calculate thermistor resistance based on voltage divider
    float R;
    if (range_idx == 0) {
        R = adc_percent * Rs[0] / (1.0f - adc_percent);
    } else {
        float Rp_eff = Rp[range_idx];
        float Rs_val = Rs[range_idx];
        float denom = Rp_eff - adc_percent * (Rs_val + Rp_eff);
        if (fabs(denom) < 1e-6f) return 0;  // Avoid division by near-zero
        R = adc_percent * Rs_val * Rp_eff / denom;
    }

    // Calculate temperature using Steinhart-Hart equation
    if (R <= 0) return 0;

    float logR = logf(R);
    float T_kelvin = 1.0f / (p0 + p1 * logR + p2 * logR * logR + p3 * logR * logR * logR);
    float T_celsius = T_kelvin - 273.15f;

    // Sanity check
    if (T_celsius < -120.0f || T_celsius > 60.0f)
        return 0;

    return T_celsius;
}
```

rs1729, 2025-05-15:
The ADC values in the "normal" temperature field and the RH-sensor look similar at this altitude.
Another possibility could be, that it is a PS20 and it has only a simple T sensor and resistor-setup similar to to TH sensor.
Then the values would be around 22C. 
However it would show 21C for the other data at 1000m as well as 12000m.
No, the ADC-T-data is too static I believe.

tushroy, 2025-05-15:
> Are these normal radiosonde soundings at 00Z/12Z the same? These two recordings show 18Z and 06Z. Or is there another radiosonde in the air at the same time? Who is launching this? Looks like the [...]

The Meteorological Department for sure. they are launching around like 00, 06, 12 and 18 hours in local time. 06 and 18 hour sondes are IMET-4. 00 and 12 are being decoded as M20.

rs1729, 2025-05-15:
> [@rs1729](https://github.com/rs1729) can you please check on following:
>
> ```
> static float get_Temp(gpx_t *gpx) {
>     // Steinhart-Hart coefficients for Shibaura PB5-41E
>     const float p0 = [...]

This gives the same values. It is the same calculations.

rs1729, 2025-05-15:
If there is no "normal" temperature ADC and no humidity data (ADC-RH=0xbb80), probably there is no humidity sensor that is heated. Perhaps there is still a T-sensor similar to the TH-sensor on the sonde, and the data in the TH field is approximately the air temperature. But that's speculation. (If it would be MCU temperature, it would probably stay above 0C for longer.)

So in such cases (T>64C) one could take the TH temperature for the JSON/sondehub output. But it would be good to confirm this a little bit.

rs1729, 2025-05-15:
> The Meteorological Department for sure. they are launching around like 00, 06, 12 and 18 hours in local time. 06 and 18 hour sondes are IMET-4. 00 and 12 are being decoded as M20.

These two M20 / possible PS20 have been launched at 18Z and 06Z, so was there also an iMet in the air, or was it this time instead of iMet?
And the M20 at 00Z and 12Z show normal temperature data?

tushroy, 2025-05-15:
> These two M20 / possible PS20 have been launched at 18Z and 06Z, so was there also an iMet in the air, or was it this time instead of iMet? And the M20 at 00Z and 12Z show normal temperature data?

- M20 / possible PS20 are always being launched at 18Z and 06Z.
- iMets are at 00Z and 12Z. and They show normal temperature data.

tushroy, 2025-05-15:
> > > > > Are these normal radiosonde soundings at 00Z/12Z the same? These two recordings show 18Z and 06Z. Or is there another radiosonde in the air at the same time? Who is launching this? Looks like the Department of Environment or Costguard, but I see other tech/science buildings on the map, and what seems to be a military airfield.
> > > > > Maybe they launch imet for PTU and additional PS20 for wind? Maybe you can track one down and recover.
> > > > 
> > > > 
> > > > The Meteorological Department for sure. they are launching around like 00, 06, 12 and 18 hours in local time. 06 and 18 hour sondes are IMET-4. 00 and 12 are being decoded as M20.
> > > 
> > > 
> > > These two M20 / possible PS20 have been launched at 18Z and 06Z, so was there also an iMet in the air, or was it this time instead of iMet? And the M20 at 00Z and 12Z show normal temperature data?
> > 
> > 
> > ```
> > * M20 / possible PS20 are always being launched at 18Z and 06Z.
> > 
> > * iMets are at 00Z and 12Z. and They show normal temperature data.
> > ```
> 
> And at 18Z / 06Z, have there been M20 launches before that showed temperature and humidity data?

Negative, the M20s never showed temperature and humidity data in my reception.

rs1729, 2025-05-15:
Ok,
perhaps it was always the pilotsonde PS20 that does not measure PTU, only wind.
Maybe you can ask the Met-Office? Or you can recover such a sonde if it lands somewhere close to you.

closed 2025-05-15 by tushroy

darksidelemm, 2025-05-17:
What was the outcome here? Do we know for sure this was a PS-20?

tushroy, 2025-05-17:
@darksidelemm Hi Mark, I couldn't verify exactly (by asking the authority or by getting the sonde on hand) whether it is M20 or PS20. But as per the launching schedules it looks like they are for wind measurement only. and consistently show same result during that time period.

rs1729, 2025-05-17:
That data looks very much like the older samples of PS20 known to me.

The TH data on the M20 is the temperature sensor of the humidity sensor.
The PS20 data shows TH data, and looking and pictures of PS20, it actually has a small temperature sensor
https://www.meteomodem.com/pilotsonde
even though it says "This GNSS sonde without Temperature and Humidity sensors". I think the temperature sensor has not the accuracy of the M20 sensors, also the resistor measuring setup is simpler (saving costs).
You can see the TH output when processing the raw data (at the end of each line):
```
./mXXmod -vv --ptu --rawhex 20250514-173933_M20_401467000.raw.txt | grep OK
Wed 2025-05-14 17:39:54.000  lat: 23.80602  lon: 90.38711  alt: 948.39   vH:  7.6  D:  27.6  vV: 3.4   SN: 310-3-01971  #  [OK]  TH:26.3C
Wed 2025-05-14 17:39:55.000  lat: 23.80610  lon: 90.38715  alt: 951.35   vH: 11.5  D:  23.0  vV: 2.9   SN: 310-3-01971  #  [OK]  TH:26.4C
[...]
Wed 2025-05-14 18:00:59.000  lat: 23.79884  lon: 90.47185  alt: 5200.77   vH: 13.6  D: 116.8  vV: 4.1   SN: 310-3-01971  #  [OK]  TH:0.2C
Wed 2025-05-14 18:01:10.000  lat: 23.79855  lon: 90.47307  alt: 5236.46   vH: 15.8  D: 115.1  vV: 2.5   SN: 310-3-01971  #  [OK]  TH:-0.1C
[...]
Wed 2025-05-14 18:24:07.000  lat: 23.76150  lon: 90.63278  alt: 9949.80   vH: 12.2  D: 126.2  vV: 4.6   SN: 310-3-01971  #  [OK]  TH:-27.6C
Wed 2025-05-14 18:24:39.000  lat: 23.75977  lon: 90.63605  alt: 10069.95   vH: 10.7  D: 108.6  vV: 5.8   SN: 310-3-01971  #  [OK]  TH:-28.5C
[...]
Wed 2025-05-14 18:41:02.000  lat: 23.72075  lon: 90.69889  alt: 13576.64   vH: 12.9  D: 107.9  vV: 5.9   SN: 310-3-01971  #  [OK]  TH:-53.7C
Wed 2025-05-14 18:41:11.000  lat: 23.72029  lon: 90.69970  alt: 13607.50   vH: 10.8  D: 110.8  vV: 4.6   SN: 310-3-01971  #  [OK]  TH:-53.9C
Wed 2025-05-14 18:42:00.000  lat: 23.71782  lon: 90.70395  alt: 13389.47   vH: 10.8  D: 121.8  vV: -22.0   SN: 310-3-01971  #  [OK]  TH:-54.9C
Wed 2025-05-14 18:42:02.000  lat: 23.71771  lon: 90.70413  alt: 13344.59   vH: 10.9  D: 122.0  vV: -21.9   SN: 310-3-01971  #  [OK]  TH:-54.8C
Wed 2025-05-14 18:42:04.000  lat: 23.71760  lon: 90.70432  alt: 13300.53   vH: 11.1  D: 119.6  vV: -21.7   SN: 310-3-01971  #  [OK]  TH:-54.6C
Wed 2025-05-14 18:43:08.000  lat: 23.71589  lon: 90.70865  alt: 11875.59   vH:  8.3  D: 122.3  vV: -21.2   SN: 310-3-01971  #  [OK]  TH:-45.2C
Wed 2025-05-14 18:44:27.000  lat: 23.71215  lon: 90.71214  alt: 10291.86   vH:  8.4  D: 143.6  vV: -19.0   SN: 310-3-01971  #  [OK]  TH:-32.9C
Wed 2025-05-14 18:44:43.000  lat: 23.71115  lon: 90.71344  alt: 9986.87   vH: 10.2  D: 121.3  vV: -18.7   SN: 310-3-01971  #  [OK]  TH:-30.6C
Wed 2025-05-14 18:44:46.000  lat: 23.71099  lon: 90.71371  alt: 9929.70   vH: 10.3  D: 127.3  vV: -18.8   SN: 310-3-01971  #  [OK]  TH:-30.2C
Wed 2025-05-14 18:44:51.000  lat: 23.71071  lon: 90.71419  alt: 9835.78   vH: 10.1  D: 119.6  vV: -18.5   SN: 310-3-01971  #  [OK]  TH:-29.6C
Wed 2025-05-14 18:44:52.000  lat: 23.71066  lon: 90.71427  alt: 9817.01   vH: 11.6  D: 130.9  vV: -18.6   SN: 310-3-01971  #  [OK]  TH:-29.4C

```
In principle, if the "normal" temperature is T > 60 (and no humidity) indicating PS20, we could use the TH data in the JSON output.
Should I modify the decoder?
(Don't know if we want to change the "type" (also in the "id"-string) though. "subtype" is from the second byte, and it is 0x20 as for regular M20.)